rust-osdev/bootloader #192: sti instruction / x86_64::instructions::interrupts::enable(); gives double fault when ran

actuallyexeon, 2021-08-21:
So I'm trying to get PIC interrupts working on the git branch of bootloader and every time I run x86_64::instructions::interrupts::enable(); it always gives me a double fault, but when run on bootimage and an older version it doesn't. Can anyone help?

actuallyexeon, 2021-08-21:
It seems like it's because I don't have APIC Timer implemented yet

phil-opp, 2021-08-22:
Yeah, it's probably because of the APIC timer. With UEFI, the legacy PIC is no longer supported, so you have to mask all of its interrupts and set up the APIC before enabling interrupts on the CPU.

actuallyexeon, 2021-08-23:
Well time to get APIC working

actuallyexeon, 2021-08-24:
Do you know of any tutorial / code that I can use for APIC?