rust-osdev/bootloader #195: No `package.metadata.bootloader.target` key found in Cargo.toml of bootloader

jamesmarva, 2021-08-22:
I add the code below in Cargo.toml
```
[package.metadata.bootloader]
target="fledgeos.json"
```
but it still happen the error 
```
Caused by:
    The `bootloader` dependency has not the right format: No `package.metadata.bootloader.target` key found in Cargo.toml of
bootloader
```

jamesmarva, 2021-08-22:
I use bootloader 0.10.6
```
[package]
name = "fledgeos-1"
version = "0.1.0"
edition = "2018"

[dependencies]
bootloader = "0.10.6"
x86_64 = "0.14.3"

[package.metadata.bootimage]
run-command = ["qemu-system-x86_64.exe", "-drive", "format=raw,file=F:/Rust-In-Action/ch11/fledgeos-1/target/fledge/debug/bootimage-fledgeos-1.bin"]

[package.metadata.bootloader]
target="fledgeos.json"
```

bjorn3, 2021-08-22:
Are you following https://os.phil-opp.com? If so you need to use bootloader 0.9 and not 0.10.

vikigenius, 2021-10-31:
Yeah, this is problematic I wanted to use 0.10 version to remove the deprecation warnings for llvm_asm! but it doesn't work due to this issue. Any clarification so as to why the entry was removed would be welcome

bjorn3, 2021-10-31:
Bootloader 0.10 works in a fundamentally different way to add support for uefi. It for example it doesn't use bootimage at all. Doesn't the latest release in the 0.9 series work without llvm_asm? Does `cargo update` work?

vikigenius, 2021-10-31:
@bjorn3 Nope, even with Cargo update the warning is still there.

calebschoepp, 2021-12-22:
@vikigenius Did you ever manage to get rid of the `llvm_asm!` warnings?

vikigenius, 2021-12-22:
Nope, but I did manage to get the latest version of bootloader working following instructions in the crate documentation.

phil-opp, 2022-01-09:
With https://github.com/rust-osdev/bootloader/pull/204, I think we can now close this.

closed 2022-01-09 by phil-opp