rust-osdev/bootloader #199: black screen on QEMU 6.0.93

jaymzee, 2021-09-18:
I've tried all of the Writing an OS in Rust posts from https://os.phil-opp.com/  and they work fine with QEMU 4.2.1 under Ubuntu 20, but when I tried running them on QEMU 6.0.93 under Windows 10, I just get a black screen.
I've tested this install of QEMU 6 with every other OS I have and they all boot fine. I'm using bootloader 0.9.19 since that seems to be the last one that is completely compatible with the configuration given in the blog posts. 
So I'm not sure where to go next.

closed 2021-09-19 by jaymzee

phil-opp, 2021-09-19:
> update: I resolved this by installing QEMU 6.1.0

Thanks for providing the update, I'm glad that it works again.

vandadnp, 2021-09-23:
I had the exact same issue as @jaymzee with QEMU 6.0.x on Windows, just a black screen, updating to QEMU 6.1.x fixed the issue for me 👍🏻