rust-osdev/bootloader #203: How to know in which mode the kernel is booted?

Narasimha1997, 2021-11-12:
As I've seen, the bootloader can boot in legacy BIOS ams UEFI modes. However, there is no way for the kernel to directly know in which mode it has been booted. This can be useful in different parts of the kernel (For example, during the initialisation of 8259 PIC as trying to enable PIC in UEFI mode double faults immediately). 

One way I've seen is to iterate over the memory map and check if the regions are of type `UnknownBios` or `UnknownUefi` and take decisions. But this isn't an idiomatic way.

The best way would be to pass a flag in `BootInfo` struct called `is_uefi` which is `true` or `false` based on the mode the kernel is booted.

DraftedDev, 2024-12-25:
You can use [BootInfo::rsdp_addr](https://docs.rs/bootloader_api/latest/bootloader_api/info/struct.BootInfo.html#structfield.rsdp_addr) if you want to find out what "mode" the kernel is running in.

bjorn3, 2024-12-25:
I think you misread the documentation of `rsdp_addr`.

> This field is None if no RSDP was found (for BIOS) or reported (for UEFI).

In other words the field will be None in the following cases:

* If BIOS is used, it is None when no RSDP is found.
* If UEFI is used, it is None when UEFI didn't report an RSDP.

`rsdp_addr` is expected to be Some on all systems that support ACPI, even when they are running under UEFI.

DraftedDev, 2024-12-26:
> I think you misread the documentation of `rsdp_addr`.
>
>
> In other words the field will be None in the following cases:
>
>     * If BIOS is used, it is None when no RSDP is found.
>
>     * If UEFI is [...]

Oh, sorry. Then it looks like a kind of boot info variables would make sense to see if you are running UEFI  or BIOS

bjorn3, 2024-12-26:
Yeah. Maybe a field which contains the address of the runtime services struct when booting as UEFI and a null pointer when booting as BIOS would work? There is currently no way to get access to the runtime services either.

ChocolateLoverRaj, 2025-02-26:
> For example, during the initialisation of 8259 PIC as trying to enable PIC in UEFI mode double faults immediately

I don't think this is the case. I was actually able to use the 8259 PIC in UEFI mode. You just have to unmask some IRQs which are masked initially on UEFI mode but not on BIOS mode.