rust-osdev/bootloader #224: [Question] Why is `cargo install bootloader --features builder` not possible?

TornaxO7, 2022-02-26:
As stated in the title: What is missing to be able to install this carte as a binary?

TornaxO7, 2022-02-26:
Ok, I see, `builder` needs the crate to since it wants to compile `uefi_bin`/`bios_bin` if you call it. Can I help you here?

TornaxO7, 2022-02-28:
Hello @bjorn3 I just saw that you replied to the issue #225 and I hope that's it's fine to ping you.
Do you know the future plans of this? Is it possible that I can help to make the `builder` installable with `cargo install builder`?

bjorn3, 2022-02-28:
@phil-opp is the author of this crate. You will probably have to ask him.

phil-opp, 2022-03-01:
I'm currently working a new 0.11 version of the crate, which should remove the need for any builder binaries. The plan is to provide the functionality as a library instead so that you can invoke it in your own way.