rust-osdev/bootloader #226: Unnecessary warnings due to inline assembly stabilized in 1.59

Programatic, 2022-03-01:


phil-opp, 2022-03-01:
Thanks for reporting! 

Since you're using `bootimage`, I assume that you're using a `v0.9` release of the `bootloader` crate. This version does indeed still tries to enable the stabilized `asm` feature. I'll fix that.

For the `v0.10` release, this was fixed in https://github.com/rust-osdev/bootloader/pull/210, which was released in `bootloader` v0.10.11. Running `cargo update -p bootloader` should fix it.

Programatic, 2022-03-01:
I was just following through the series you have on your website, so I assumed that it was not updated yet when I posted this issues, my bad! I tried using v0.10, but couldn't get it working in the few minutes I had spent on it. Is there plans to update the blog to use the newest version, or is this an area that needs help?

phil-opp, 2022-03-01:
Yeah, the blog does not work with bootloader v0.10 yet. There were some substantial changes in v0.10 (e.g. UEFI support), so parts of the blog require a rewrite. However, I'm not fully satisfied with the build system of v0.10 yet, so I'm currently working on a new v0.11 version, which I will then use for the next iteration of the blog.

This means that things are in a weird state right now, where both the v0.9 and v0.10 versions of the bootloader crate are actively updated. Sorry for the confusion! I created a PR to fix these warnings in https://github.com/rust-osdev/bootloader/pull/227.

phil-opp, 2022-04-16:
Fixed in #227

PR #227 (2022-03-01): Remove the `asm` and `global_asm` features, which are now stable
Fixes #226
closed 2022-04-16 by phil-opp