rust-osdev/bootloader #236: 'kernel mapping failed'

Veliania, 2022-04-29:
I'm not sure what to do about this or if this is the right place but whenever I try to run the very barest of what I understand should work in terms of code for a kernel
```
#![no_std]
#![no_main]

use core::panic::PanicInfo;

#[no_mangle]
pub extern "C" fn _start() -> ! {
    loop {}
}

#[panic_handler]
fn panic(_panic_info: &PanicInfo) -> ! {
    loop {}
}
```
But no matter what it wont work and it throws an error saying ```panicked at 'kernel mapping failed: Mapping(PageAlreadyMapped(PhysFrame[4KiB](0x401000)))', src\main.rs:285:6```
I'm building using x86_64-unkown-none as the target and using the BootImage command to build it

Freax13, 2022-06-26:
The `bootimage` command uses version 0.9 of the bootloader which doesn't support `x86_64-unkown-none`. You should use a [custom target spec](https://os.phil-opp.com/minimal-rust-kernel/#target-specification) instead.

phil-opp, 2023-01-02:
I think this is answered.

closed 2023-01-02 by phil-opp