rust-osdev/bootloader #242: plans for supporting aarch64 architecture

oneforalone, 2022-07-20:
I'm using bootloader on m1 chip. Is there any plans for supoorting this architecture. Or is there any alternatives for arm64 architecture?

bjorn3, 2022-07-20:
I think it would be doable to add AArch64 support to the UEFI version of the bootloader in https://github.com/rust-osdev/bootloader/pull/232. As for native M1 support, I think you should use m1n1 with U-Boot as payload to provide a UEFI interface which can then load bootloader once it supports AArch64 UEFI.

andrewdavidmackenzie, 2026-09-23:
Checking in again to see if aarch64 support could land sometime?
Among other arch (x86_64 and riscv) I am targetting aarch64 (mainly Pi4 and above) with my kernel and it would be "nice" to standardize on bootloader (currently using on x86).

If not likely to happen soon, best just to break it for me and I'll look elsewhere for that arch.

thanks!

Freax13, 2026-09-23:
> If not likely to happen soon, best just to break it for me and I'll look elsewhere for that arch.

I am not aware of any ongoing work to add support for aarch64.