rust-osdev/bootloader #246: No serial print

Skyslycer, 2022-07-27:
Hi again, I want to output some text, so I searched this repository and found [serial.rs](https://github.com/rust-osdev/bootloader/blob/main/examples/test_framework/src/serial.rs) in the test framework. So I copied it and ran https://paste.skyslycer.de/fvkbX.

The only thing _actually_ showing up tho is the startup text. https://upload.skyslycer.de/qemu-system-x86_64_ios9o2Xd0E.png

Any ideas?

phil-opp, 2022-07-28:
The serial text is not printed to the screen. You need to tell QEMU how to handle it. For example, you could pass `-serial stdio` to QEMU to print the serial output to the stdout of the terminal.

Skyslycer, 2022-07-28:
> The serial text is not printed to the screen. You need to tell QEMU how to handle it. For example, you could pass `-serial stdio` to QEMU to print the serial output to the stdout of the terminal.

Would that work with a physical machine?

phil-opp, 2022-07-28:
On a real machine you run the OS directly on the hardware, so you don't use QEMU. The [serial port](https://en.wikipedia.org/wiki/Serial_port) is an actual physical interface there, but you probably won't find it on modern computers anymore. You might be able to use an adapter, but I don't have any experience with that.

Skyslycer, 2022-07-28:
How do I print it to the screen then? Frame buffers? Don't see any good examples/documentation.

bjorn3, 2022-07-28:
Are you using BIOS or UEFI? For BIOS you can either use VGA text mode, or a VGA or VESA framebuffer. For UEFI you have to use a GOP framebuffer. Versions of bootloader from before UEFI support was added, have VGA text mode set up by default. The versions with UEFI support have a VGA or GOP framebuffer set up and pass you a pointer to it in the BootInfo struct.

Skyslycer, 2022-07-28:
> Are you using BIOS or UEFI? For BIOS you can either use VGA text mode, or a VGA or VESA framebuffer. For UEFI you have to use a GOP framebuffer. Versions of bootloader from before UEFI support was [...]

Is there anything universal or a library?

bjorn3, 2022-07-28:
This is the code the bootloader crate uses for logging: https://github.com/rust-osdev/bootloader/blob/ac46d0455b41c11e5d316348d068df1c495ce0af/src/binary/logger.rs It uses the noto_sans_mono_bitmap crate for rendering the Noto Sans Mono font to a framebuffer.

Skyslycer, 2022-08-07:
> This is the code the bootloader crate uses for logging: https://github.com/rust-osdev/bootloader/blob/ac46d0455b41c11e5d316348d068df1c495ce0af/src/binary/logger.rs It uses the noto_sans_mono_bitmap crate for rendering the Noto Sans Mono font to a framebuffer.

I can't seem to be able to use that correctly. For testing purposes, I just used the class and added the required crates. Then I used `log::info!()` to try to write something. It didn't.
https://upload.skyslycer.de/firefox_eFXZxYpYog.png This is everything I see. Not my desired output, in fact, no output from the kernel itself.

bjorn3, 2022-08-07:
Are you trying to use the `log` crate? That won't work. You need to use the `Logger` type defined in the file I linked.

Skyslycer, 2022-08-07:
I'll try that.