rust-osdev/bootloader #251: [Need Help] Interrupt return error with gdt set

AmberCrafter, 2022-08-23:
I got a double fault exception at interrupt return with gdt setted. And I try to figure out what happened, but feel confused. Below is my testing results

Cross test:
1. Same gdt and interrupt setting (same as glob_os 2ed edition) with
 - bootlaoder(v0.9) + bootimage: work
 - bootloader(v0.10.12): not work

2. Base on [blog_os 2ed edition post-07](https://github.com/phil-opp/blog_os/tree/post-07) and modifying it to use newer bootlaoder(v0.10.12), still has same exception.

> Note
>1. Same kind of error ocurr at the brackpoint/timer interrupt return, but different in mechine code.
>2. The brackpoint exception work fine without gdt setting. (Disable src/lib.rs `library::blog_gdt::init();`)

---
Environment and Dependency
github codespace
rust edition: 2021
bootloader: 0.10.12
x86_64: 0.14.10

github info:
branch: interrupt
tag: [interrupt_return_double_fault](https://github.com/AmberCrafter/rustos/tree/interrupt_return_double_fault)

---
## Log

### Qemu -d int dump

Command
```bash
> cargo krun
```

<details>
    <summery>Qemu outputs</summery>
    Qemu args: ["-drive", "format=raw,file=/workspaces/rustos/target/x86_64-rustos-none/debug/boot-bios-rustos.img", "-d", "int", "--no-reboot", "-s", "-display", "none", "-serial", "mon:stdio"]

    ```
    Finished test [unoptimized + debuginfo] target(s) in 0.04s
    Running tests/interrupt.rs (target/x86_64-rustos-none/debug/deps/interrupt-b5aa4371b9e2d1be)
    Finished dev [unoptimized + debuginfo] target(s) in 0.01s
    Running `target/debug/image_maker /workspaces/rustos/target/x86_64-rustos-none/debug/deps/interrupt-b5aa4371b9e2d1be`
    CommandArgs { inner: ["-drive", "format=raw,file=/workspaces/rustos/target/x86_64-rustos-none/debug/deps/boot-bios-interrupt-b5aa4371b9e2d1be.img", "-d", "int", "--no-reboot", "-s", "-display", "none", "-serial", "mon:stdio"] }
    SMM: enter
    EAX=00000001 EBX=00000000 ECX=02000000 EDX=02000628
    ESI=0000000b EDI=02000000 EBP=000f246b ESP=00006ce0
    EIP=000eae06 EFL=00000002 [-------] CPL=0 II=0 A20=1 SMM=0 HLT=0
    ES =0010 00000000 ffffffff 00cf9300 DPL=0 DS   [-WA]
    CS =0008 00000000 ffffffff 00cf9b00 DPL=0 CS32 [-RA]
    SS =0010 00000000 ffffffff 00cf9300 DPL=0 DS   [-WA]
    DS =0010 00000000 ffffffff 00cf9300 DPL=0 DS   [-WA]
    FS =0010 00000000 ffffffff 00cf9300 DPL=0 DS   [-WA]
    GS =0010 00000000 ffffffff 00cf9300 DPL=0 DS   [-WA]
    LDT=0000 00000000 0000ffff 00008200 DPL=0 LDT
    TR =0000 00000000 0000ffff 00008b00 DPL=0 TSS32-busy
    GDT=     000f62c0 00000037
    IDT=     000f62fe 00000000
    CR0=00000011 CR2=00000000 CR3=00000000 CR4=00000000
    DR0=0000000000000000 DR1=0000000000000000 DR2=0000000000000000 DR3=0000000000000000 
    DR6=00000000ffff0ff0 DR7=0000000000000400
    CCS=00000080 CCD=00000001 CCO=LOGICB  
    EFER=0000000000000000
    SMM: after RSM
    EAX=00000001 EBX=00000000 ECX=02000000 EDX=02000628
    ESI=0000000b EDI=02000000 EBP=000f246b ESP=00006ce0
    EIP=000eae06 EFL=00000002 [-------] CPL=0 II=0 A20=1 SMM=0 HLT=0
    ES =0010 00000000 ffffffff 00c09300 DPL=0 DS   [-WA]
    CS =0008 00000000 ffffffff 00c09b00 DPL=0 CS32 [-RA]
    SS =0010 00000000 ffffffff 00c09300 DPL=0 DS   [-WA]
[... 495 lines cut ...]
```

<details>
    <summery>objdump outputs</summery>

    ```
    (empty)
    ```
</details>
According

---

I'm a newer on os kernel, thus i can't ensure my thinking correct or not:

```
    check_exception old: 0xffffffff new 0xd
        1: v=0d e=0010 i=0 cpl=0 IP=0008:0000000000211b58 pc=0000000000211b58 
```
Base on [stackoverflow](https://stackoverflow.com/questions/62828313/general-protection-fault-when-trying-to-sti) analysis:
According to the Qemu dump info
exception number show 0xd which mean run into the [general protection fault](https://wiki.osdev.org/Exceptions#General_Protection_Fault)
and e=0010 show the error is trigger by GDT 2ed entry

This make me confuse why failed at this point.

phil-opp, 2022-08-23:
Sorry that this is still not documented. I'm currently working on a new `v0.11` version in #232, which will have proper migration guides that document such subtle changes.

AmberCrafter, 2022-08-24:
Sorry for long time no response. I try to understand what are #196 and #190 mention, and trace the variety of the registers in these days. Here is my [results](https://github.com/AmberCrafter/rustos/blob/solve_interrupt_return_double_fault/documents/BugReport/experiment_iretq_64bit.md).
Thank you, @phil-opp, provides these clear documents and crate, it's vary easy to understand how do the os kerenl work, and i've learned alot from here!!

closed 2022-08-24 by AmberCrafter