rust-osdev/bootloader #257: Examples run very slow in Apple M1

zhengcan, 2022-09-15:
I'm running the examples in Apple M1 by using `cargo krun`.
And I've upgrade qemu to the latest version (v7.1.0).

The examples compile as usual, but run very slow.
The qemu window show nothing and stay freeze for several minutes, and turn gray later.

bjorn3, 2022-09-15:
This is expected as this crate is an x86_64 bootloader. If you try to run it on another architecture like AArch64, qemu will use software based virtualization of an x86_64 cpu, which is significantly slower than hardware accelerated virtualization as possible on real x86_64 hardware.

Freax13, 2022-09-15:
> This is expected as this crate is an x86_64 bootloader. If you try to run it on another architecture like AArch64, qemu will use software based virtualization of an x86_64 cpu, which is significantly slower than hardware accelerated virtualization as possible on real x86_64 hardware.

We don't run the virtual machines with `-enable-kvm`. Does qemu still gain from running on the same platform even when kvm is not enabled?

bjorn3, 2022-09-15:
I thought we enabled kvm. But because that is not the case, please disregard my previous comment.

zhengcan, 2022-09-17:
> I thought we enabled kvm. But because that is not the case, please disregard my previous comment.

😄 I've tried to launch ubuntu by using qemu-system-x86_64. It runs very quick.