rust-osdev/bootloader #258: bootloader debug symbols

hawkw, 2022-09-17:
Hi, I'm attempting to use the `bootloader` crate with a debugging workflow where a kernel is run in QEMU using QEMU's support for `gdb`'s remote debugging protocol. This workflow works fine once the kernel code is entered, as it's possible to load the original, un-`strip`ped kernel binary into gdb to access its debug symbols. However, the `bootloader` build script doesn't appear to have a way to output debug symbols for the bootloader itself, so it's not possible for gdb to find function boundaries etc while still in the bootloader.

Would it be possible to add an option to get the build script to emit the bootloader's debug symbols? Building in release mode is probably still necessary, but the bootloader could be built in release mode with debug symbols enabled, and then `strip`ped to remove the debug symbols from the binary that's linked into the final output binary. That would help a lot when debugging issues that (appear) to occur before the jump into the kernel itself.

hawkw, 2022-09-17:
I'm happy to implement something like this myself, but I might need some guidance from a maintainer.

Freax13, 2022-09-18:
Assuming you're using the bios bootloader the unstripped bootloader should be in `target/x86_64-bootloader/release/bios` relative the the `Cargo.toml` of the kernel.
So for example for the `default_settings` tests:
```
$ file tests/test_kernels/default_settings/target/x86_64-bootloader/release*
bios:                          ELF 64-bit LSB executable, x86-64, version 1 (SYSV), statically linked, with debug_info, not stripped
bios.d:                        ASCII text, with very long lines (1521)
boot-bios-basic_boot.img:      DOS/MBR boot sector
boot-bios-check_boot_info.img: DOS/MBR boot sector
boot-bios-should_panic.img:    DOS/MBR boot sector
build:                         directory
deps:                          directory
examples:                      directory
incremental:                   directory
```

On the new [0.11 branch](https://github.com/rust-osdev/bootloader/pull/232) those files are in `target/debug/build/bootloader-<some-hash>/out/bin`:
```
$ file target/debug/build/bootloader-c0657fd86cef9512/out/bin/*
bootloader-x86_64-bios-boot-sector:     ELF 32-bit LSB executable, Intel 80386, version 1 (SYSV), statically linked, not stripped
bootloader-x86_64-bios-boot-sector.bin: DOS/MBR boot sector
bootloader-x86_64-bios-stage-2:         ELF 32-bit LSB executable, Intel 80386, version 1 (SYSV), statically linked, not stripped
bootloader-x86_64-bios-stage-2.bin:     data
bootloader-x86_64-bios-stage-3:         ELF 32-bit LSB executable, Intel 80386, version 1 (SYSV), statically linked, with debug_info, not stripped
bootloader-x86_64-bios-stage-3.bin:     data
bootloader-x86_64-bios-stage-4:         ELF 64-bit LSB executable, x86-64, version 1 (SYSV), statically linked, with debug_info, not stripped
bootloader-x86_64-bios-stage-4.bin:     data
bootloader-x86_64-uefi.efi:             PE32+ executable (EFI application) x86-64, for MS Windows
```

hawkw, 2022-09-20:
@Freax13 yup, I found the unstripped bootloader binary and loaded it into GDB using `add-symbol-file`, and it does work! So we can probably close this issue.

It might be nice if the symbols were stripped only after linking, so that we could produce a single binary that would contain both kernel and bootloader symbols, to simplify the debugging workflow, but this isn't a major issue.

closed 2022-09-20 by hawkw