rust-osdev/bootloader #262: panic when both `map-physical-memory` and `dynamic-range-start` are enabled

hawkw, 2022-09-22:
Trying to test out a higher half kernel using PR #229, the bootloader panics when the `map-physical-memory`, `map-framebuffer`, and `dynamic-range-start` configurations are all present.

I have the following configurations in my `Cargo.toml`:
```toml
[package.metadata.bootloader]
map-physical-memory = true
# the kernel is mapped into the higher half of the virtual address space.
dynamic-range-start = "0xFFFF_8000_0000_0000"

[patch.crates-io]
# patch `bootloader` to use https://github.com/rust-osdev/bootloader/pull/229
# for dynamic range address configuration support.
#
# remove this patch once rust-osdev/bootloader#229 makes it into a release
# version.
bootloader = { git = "https://github.com/rust-osdev/bootloader", rev = "ac46d0455b41c11e5d316348d068df1c495ce0af" }
```

I get the following panic:
[image]

hawkw, 2022-09-22:
When the `map-page-table-recursively` setting is also `true`, there's a different panic while setting up the recursive mapping:
[image]

```toml
[package.metadata.bootloader]
map-physical-memory = true
map-page-table-recursively = true
dynamic-range-start = "0xFFFF_8000_0000_0000"
```

Freax13, 2022-09-22:
I feel like this is also related to https://github.com/rust-osdev/bootloader/issues/259. I can reproduce this on qemu 7.1, but not on 7.0.

Freax13, 2022-09-22:
https://github.com/rust-osdev/bootloader/blob/ac46d0455b41c11e5d316348d068df1c495ce0af/src/binary/level_4_entries.rs#L172-L174
This allocates a level 4 entry which spans 2^39 bytes irregardless of the requested size. QEMU's reported memory regions at 1TB are larger than that (2^40 bytes) which results the physical memory mapping overlapping into other regions.

Freax13, 2022-09-22:
`dynamic-range-start` is not actually required for the crash, only `map-physical-memory`.

hawkw, 2022-09-22:
> `dynamic-range-start` is not actually required for the crash, only `map-physical-memory`.

Ah, yeah, you're right &mdash; it turns out I can reproduce this without the dynamic range configuration, and it's having `map-physical-memory` enabled that's the culprit. My bad!

This is definitely related to #259, then.

hawkw, 2022-09-22:
I can close this issue as a duplicate of #259 if you like?

Freax13, 2022-09-22:
Sure

PR #263 (2022-09-25): fix `get_free_address` for large sizes (0.10)
This pr fixes `get_free_address` for large sizes (more than 2^39).

Part of #259
Closes #262
PR #264 (2022-09-25): allow allocating more than one level 4 entry (0.9)
This pr fixes large physical memory mappings (> 512GiB).

Part of #259
Closes #262
Closes https://github.com/phil-opp/blog_os/issues/1138
closed 2022-09-22 by hawkw