rust-osdev/bootloader #270: `bootloader` dependency has not the right format: No `package.metadata.bootloader.target` key found in Cargo.toml of bootloader

saivishwak, 2022-10-08:
Finished dev [unoptimized + debuginfo] target(s) in 4.80s
Error: An error occured while trying to build the bootloader: The `bootloader` dependency has not the right format: No `package.metadata.bootloader.target` key found in Cargo.toml of bootloader

(If you're using the official bootloader crate, you need at least version 0.5.1)

Caused by:
    The `bootloader` dependency has not the right format: No `package.metadata.bootloader.target` key found in Cargo.toml of bootloader

    (If you're using the official bootloader crate, you need at least version 0.5.1)

saivishwak, 2022-10-08:
With latest rust nightly the issue is seen.

bjorn3, 2022-10-08:
Can you post the `Cargo.toml` file you are using?

saivishwak, 2022-10-08:
[package]
name = "rob_os"
version = "0.1.0"
edition = "2021"

[dependencies]
bootloader = "0.10.13"

[package.metadata.bootimage]
build-command = ["build"]

bjorn3, 2022-10-08:
If you are following https://os.phil-opp.com you will have to use bootloader version 0.9, not 0.10.

saivishwak, 2022-10-08:
Oh Okay, Thanks a lot. Was getting a linker error in 0.9.8 in latest nightly so switched to 0.10.13.

closed 2022-10-08 by saivishwak

phil-opp, 2022-10-10:
See https://github.com/rust-osdev/bootloader/issues/271 for details on that linker error. It's a regression in Rust and should be resolved in a few days.