rust-osdev/bootloader #271: linking fails on Rust nightly-2022-10-08

hawkw, 2022-10-09:
After updating the Rust toolchain to `nightly-2022-10-08`, attempting to build a kernel using the BIOS version of `bootloader` v0.10.13 fails with a number of linker errors similar to the following (for various symbols defined by the bootloader):
```
rust-lld: error: undefined symbol: _rest_of_bootloader_start_addr
          >>> referenced by bios.98c44c40-cgu.3
          >>>               /home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.3.rcgu.o:(.boot-first-stage+0x60)
          >>> referenced by bios.98c44c40-cgu.3
          >>>               /home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.3.rcgu.o:(.boot-first-stage+0x6D)
          >>> referenced by bios.98c44c40-cgu.3
          >>>               /home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.3.rcgu.o:(.boot-first-stage+0x89)
```

It seems like the `rust-lld` behavior has changed somehow.

It would probably be useful to determine which nightly introduced this regression, in order to pinpoint the change in linker behavior. We could do this using the [`cargo bisect-rustc` tool](https://github.com/rust-lang/cargo-bisect-rustc), but, unfortunately, that tool doesn't work on my machine, so if someone else has the time to do a bisect run, that could be very helpful. I can offer `nightly-2022-07-31` as a known-good nightly (it's the version my kernel is currently pinned to), but I imagine there's almost certainly a newer working nightly...

## Steps to reproduce:

```shell
:; rustup toolchain install nightly-2022-10-09 \
    && cd examples/basic \
    && cargo +nightly-2022-10-09 kimage
```

<details>
<summary>Complete output of <code>cargo kimage</code> in <code>examples/basic</code> kernel:</summary>

```shell
:# eliza at noctis in bootloader/examples/basic on  main [$✘?] ⚙️ v1.66.0-nightly
:; cargo +nightly-2022-10-09 kimage
   Compiling compiler_builtins v0.1.79
   Compiling core v0.0.0 (/home/eliza/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/core)
   Compiling bootloader v0.10.12 (/home/eliza/Code/bootloader)
   Compiling rustc-std-workspace-core v1.99.0 (/home/eliza/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/src/rust/library/rustc-std-workspace-core)
   Compiling basic_example v0.1.0 (/home/eliza/Code/bootloader/examples/basic)
    Finished dev [unoptimized + debuginfo] target(s) in 7.94s
     Running `cargo run --package simple_boot -- target/x86_64-custom/debug/basic_example`
   Compiling json v0.12.4
   Compiling bootloader-locator v0.0.4
   Compiling locate-cargo-manifest v0.2.2
   Compiling simple_boot v0.1.0 (/home/eliza/Code/bootloader/examples/basic/simple_boot)
    Finished dev [unoptimized + debuginfo] target(s) in 0.65s
     Running `target/debug/simple_boot target/x86_64-custom/debug/basic_example`
error: linking with `rust-lld` failed: exit status: 1
  |
  = note: "rust-lld" "-flavor" "gnu" "/run/user/1000/rustcMXO3Et/symbols.o" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.0.rcgu.o" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.1.rcgu.o" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.10.rcgu.o" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.11.rcgu.o" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.12.rcgu.o" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.13.rcgu.o" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.14.rcgu.o" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.15.rcgu.o" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.2.rcgu.o" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.3.rcgu.o" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.4.rcgu.o" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.5.rcgu.o" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.6.rcgu.o" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.7.rcgu.o" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.8.rcgu.o" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.9.rcgu.o" "--as-needed" "-L" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps" "-L" "/home/eliza/Code/bootloader/examples/basic/target/release/deps" "-L" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/build/bootloader-1e7d0531e3e8bfcb/out" "-L" "/home/eliza/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-bootloader/lib" "-Bstatic" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/librsdp-f0994b510951e1cc.rlib" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/libbootloader-72e232d9fbdb1766.rlib" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/liblog-edef54c6f00d1854.rlib" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/libcfg_if-5d3e665bb59dc616.rlib" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/libspinning_top-ed2aa1630c31090d.rlib" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/liblock_api-e44b8cbd42f4efd0.rlib" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/libscopeguard-f52e6e2592823134.rlib" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/libnoto_sans_mono_bitmap-d9ee7d27604ff308.rlib" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/libconquer_once-9c405c840f71c06b.rlib" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/libxmas_elf-8c1144fae3f3a670.rlib" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/libzero-c586f84297d25602.rlib" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/librand-400a474582969ba9.rlib" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/libraw_cpuid-db447fdebf7b5197.rlib" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/librand_chacha-04264d589c30f845.rlib" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/libppv_lite86-37cedba29a39dd97.rlib" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/librand_core-fb604127c04776b0.rlib" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/libx86_64-19c2b6e654c165ce.rlib" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/libvolatile-d4f8663cc983ac90.rlib" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/libbitflags-877a0c77899c9e76.rlib" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/libbit_field-cc8cdf0c0596fe6e.rlib" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/libusize_conversions-d85a04799a0c7be3.rlib" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/librustc_std_workspace_core-b8a2fd09bcb24600.rlib" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/libcore-47af85afbcb706fe.rlib" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/libcompiler_builtins-1d57e3c7f4a6f9fd.rlib"  [... cut ...]
             at ./simple_boot/src/main.rs:24:16
   3: core::ops::function::FnOnce::call_once
             at /rustc/8796e7a9cfd4c5c4f1de15ec1c53994ddf288665/library/core/src/ops/function.rs:251:5
note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
```

</details>

<details>

<summary><code>rustc --version</code> output:</summary>

```shell
:# eliza at noctis in bootloader/examples/basic on  main [$✘?] ⚙️ v1.66.0-nightly
:; rustc --version
rustc 1.66.0-nightly (8796e7a9c 2022-10-08)
```

</details>

clstatham, 2022-10-09:
This has been driving me insane all day. I don't remember why, but something prompted me to update my nightly installation of Rust for my OS project this morning and after I did so my project stopped linking properly, even though it was working fine last night. I can provide that nightly-2022-10-07 is likely the most recent working version. I just tested it myself and everything built fine! :)

hawkw, 2022-10-09:
Taking a closer look at the output, I notice that it looks like the command line for `rust-lld` was the following (newlines added for clarity):
```shell
"rust-lld" \
    "-flavor" "gnu" \
    "/run/user/1000/rustcMXO3Et/symbols.o" \
    "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.0.rcgu.o" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.1.rcgu.o" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.10.rcgu.o" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.11.rcgu.o" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.12.rcgu.o" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.13.rcgu.o" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.14.rcgu.o" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.15.rcgu.o" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.2.rcgu.o" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.3.rcgu.o" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.4.rcgu.o" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.5.rcgu.o" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.6.rcgu.o" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.7.rcgu.o" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.8.rcgu.o" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb.bios.98c44c40-cgu.9.rcgu.o" \
     "--as-needed" \
     "-L" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps" \
     "-L" "/home/eliza/Code/bootloader/examples/basic/target/release/deps" \
     "-L" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/build/bootloader-1e7d0531e3e8bfcb/out" \
     "-L" "/home/eliza/.rustup/toolchains/nightly-x86_64-unknown-linux-gnu/lib/rustlib/x86_64-bootloader/lib" \
     "-Bstatic" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/librsdp-f0994b510951e1cc.rlib" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/libbootloader-72e232d9fbdb1766.rlib" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/liblog-edef54c6f00d1854.rlib" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/libcfg_if-5d3e665bb59dc616.rlib" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/libspinning_top-ed2aa1630c31090d.rlib" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/liblock_api-e44b8cbd42f4efd0.rlib" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/libscopeguard-f52e6e2592823134.rlib" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/libnoto_sans_mono_bitmap-d9ee7d27604ff308.rlib" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/libconquer_once-9c405c840f71c06b.rlib" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/libxmas_elf-8c1144fae3f3a670.rlib" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/libzero-c586f84297d25602.rlib" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/librand-400a474582969ba9.rlib" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/libraw_cpuid-db447fdebf7b5197.rlib" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/librand_chacha-04264d589c30f845.rlib" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/libppv_lite86-37cedba29a39dd97.rlib" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/librand_core-fb604127c04776b0.rlib" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/libx86_64-19c2b6e654c165ce.rlib" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/libvolatile-d4f8663cc983ac90.rlib" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/libbitflags-877a0c77899c9e76.rlib" \
     "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/libbit_field-cc8cdf0c0596fe6e.rlib" \
[... cut ...]
     "-o" "/home/eliza/Code/bootloader/examples/basic/target/x86_64-bootloader/release/deps/bios-f9dd4f582be5a1cb" \
     "--gc-sections"
```
Critically, the linker command does *not* appear to contain the `--script=linker.ld` flag specified in the `x86_64-bootloader.json` targetfile.

If the linker was not passed the correct linker script, that _would_ explain why these symbols are missing, as they're generated by the linker script. I wonder if something changed in how Rust determines linker arguments after `nightly-2022-10-07`?

hawkw, 2022-10-09:
Ahh, looking at the Rust issue tracker, I found PR rust-lang/rust#102836, which fixes a regression that broke the parsing of `pre-link-args` in custom targets: https://github.com/rust-lang/rust/pull/101988#issuecomment-1272407248. I bet that's our problem!

hawkw, 2022-10-09:
Hopefully this will be fixed upstream soon! In the meantime, I don't think there's anything `bootloader` can do about it besides advise pinning to `nightly-2022-10-07` or earlier.

phil-opp, 2022-10-10:
Thanks a lot for reporting and debugging this! Looks like it will be fixed by https://github.com/rust-lang/rust/pull/102836. Until then, using an older nightly is probably the best workaround.

phil-opp, 2022-10-14:
The latest nightly should fix this.

phil-opp, 2022-10-23:
Closing, as this should be fixed.

closed 2022-10-23 by phil-opp