rust-osdev/bootloader #276: Confusion about lgdt

jiangbo, 2022-11-10:
in src/asm/stage_1.s

```asm
...
   lgdt [gdt32info]
...
gdt32info:
   .word gdt32_end - gdt32 - 1  # last byte in table
   .word gdt32  
...
```
Does this mean that gdt is only two words length? But some sources say that gdtr should have 48 bits:
https://docs.oracle.com/cd/E19455-01/806-3773/instructionset-82/index.html

Freax13, 2022-11-10:
The gdtr is 5 bytes for 16 bit mode, the two `.words` are indeed not enough, there should be an additional byte. In this case this works because the GDT is placed directly after the `gdt32info` and so the first byte of the zero entry of the GDT acts also happens to overlap with the last byte of `gdt32info`.

jiangbo, 2022-11-10:
Oh, it uses the first byte of the zero entry of the GDT.

closed 2022-11-10 by jiangbo