rust-osdev/bootloader #277: UEFI v 0.11 is failing to detect the `bootloader_api` version

kennystrawnmusic, 2022-11-12:
In the process of attempting to port my kernel to version 0.11 of the bootloader crate and managed to get it to successfully compile — ah, but when I try to run it I get this (using version 0.1.0-alpha.0 of bootloader_api):
[image]

Note that instead of stating the version as the correct `0.1.0-alpha.0` it's stating the version as `()` (the unit type). Why would this be?

Freax13, 2022-11-12:
The function for deserializing the configuration doesn't have a proper error type yet:
https://github.com/rust-osdev/bootloader/blob/next/api/src/config.rs#L151

phil-opp, 2022-11-13:
Thanks for reporting! I created a fix in #280.

PR #280 (2022-11-13): Add basic error strings for deserialization errors
Fixes #277
closed 2022-11-13 by phil-opp