rust-osdev/bootloader #281: BIOS booting: Consider framebuffer config of kernel ELF

phil-opp, 2022-11-13:
Right now the BIOS implementation does not consider the framebuffer configuration of the kernel. Instead, it just defaults to a hardcoded resolution.

To fix this, we would need to parse the kernel ELF file already in the real mode stages, which could be challenging because of the memory limits in real mode. Alternatively, we could copy the framebuffer settings to a small text file and place this file alongside the kernel executable on the FAT partition.

Freax13, 2022-11-28:
We could load the file in long mode, extract the frame buffer configuration, drop down to real mode, do the VBE calls, and switch back to long mode.

phil-opp, 2022-12-29:
Good idea! I never tried going back to real mode from long mode though, so I'm not sure if it will work.