rust-osdev/bootloader #284: Frame allocation fails when creating a kernel stack larger than 66MB

kennystrawnmusic, 2022-11-18:
When setting `CONFIG.mappings.kernel_stack_size` to a larger value than 66 megabytes, this happens: 
[image]

Not sure why this is, but definitely something that needs to be looked into, especially as kernels increase in complexity and need proper page alignment for an ever-increasing number of tasks.

Freax13, 2022-11-18:
That's an out of memory error, how much RAM did you assign the VM?

kennystrawnmusic, 2022-11-18:
Alright, adding `-m 4G` to the list of arguments that my runner crate passes to QEMU did increase the point at which this occurs. Closing

closed 2022-11-18 by kennystrawnmusic