rust-osdev/bootloader #285: Support position independent code in higher half

phil-opp, 2022-11-24:
When a PIC kernel is located in higher half, we currently try to allocate 33554177 entries in the level 4 page table, which of course fails. This is the reason that we currently use the `static` relocation model for our higher half test:

https://github.com/rust-osdev/bootloader/blob/4e2b8d7e4b4fad3c7aefd7f8174f86ec2119ecc7/Cargo.toml#L87

I think the issue is in these lines:

https://github.com/rust-osdev/bootloader/blob/4e2b8d7e4b4fad3c7aefd7f8174f86ec2119ecc7/common/src/load_kernel.rs#L64-L75

Looks like we only consider the end address for the size calculation, without taking the start address into account. I think we should be able to fix this by using the difference between the max and min address as size instead.

phil-opp, 2022-11-24:
cc @Freax13

phil-opp, 2022-11-24:
I started to look into this in <https://github.com/rust-osdev/bootloader/tree/higher-half-pic>. Not sure if my approach is right, though. The current error is "the relocation table must end in the elf file".

Freax13, 2022-11-24:
Unless I'm missing something, specifying a image base address for a PIC kernel is meaningless because they're supposed to be able to be relocated anywhere. Instead the `dynamic_start` config value should be used (and already is for our higher half kernel test). 
Removing those two lines should do the trick:
https://github.com/rust-osdev/bootloader/blob/17ba81613b2feb55ca9d7be67a8923cb703d4511/Cargo.toml#L96-L97

Freax13, 2022-11-24:
I'll take a look at your approach of changing how we calculate the size later today.

phil-opp, 2022-11-24:
Thanks a lot! This issue is not that urgent, so no hurry. 

> Unless I'm missing something, specifying a image base address for a PIC kernel is meaningless because they're supposed to be able to be relocated anywhere.

I guess there are different possible ways to handle this. For example, we could try to default to the image base address if it's still free when ASLR is disabled. Or we could treat it as a fallback for the `dynamic_start` if it isn't specified. Or we just ignore it.

Either way, we should not panic on PIC kernels with a higher half base address. It should be possible to change the relocation mode without adjusting all addresses in a linker script.

Freax13, 2022-11-24:
> I guess there are different possible ways to handle this. For example, we could try to default to the image base address if it's still free when ASLR is disabled. Or we could treat it as a fallback for the `dynamic_start` if it isn't specified. Or we just ignore it.

There is some precedence for treating the base address as the default: glibc used to treat the base address as the preferred address for PIEs when the `LD_USE_LOAD_BIAS` environment variable was set to `1`. This was used in conjunction with [prelinking](https://linux.die.net/man/8/prelink) which is a process that assigns static load addresses to shared libraries and PIEs at build time. Prelinking isn't used anymore which is my `LD_USE_LOAD_BIAS` was removed from glibc earlier this year. 

> Either way, we should not panic on PIC kernels with a higher half base address. It should be possible to change the relocation mode without adjusting all addresses in a linker script.

Agreed.

Freax13, 2022-11-24:
> The current error is "the relocation table must end in the elf file".

We currently assume that the address given by the `RELA` entry is an offset into the file. This is incorrect, it's actually a virtual address pointing to mapped memory. The fix for this isn't entirely trivial because we need to translate the virtual address to a physical address (or in other words we need to translate from the kernel's address space to the bootloader's address space). I'm working on a fix.

PR #289 (2023-01-09): support higher half position independent kernels
This pr adds support for loading position independent kernels with a base address other than 0.

Currently the base address is ignored. In the future, we could try to map the kernel at it's base address if it's non zero. Doing this right now would be difficult because we don't have accurate book keeping on what addresses have already been allocated (See #239). I propose we take another look at this once finer granular ASLR is implemented (finer granular ASLR requires more accurate book keeping).

Closes #285
closed 2023-01-09 by phil-opp