rust-osdev/bootloader #286: Building recompiles bootloader every build, resulting in extreme build times

BigBadE, 2022-11-26:
Following the docs and making a simple project, a basic OS takes over 2 minutes to compile.

How to replicate:
- Pull https://github.com/BigBadE/RustOS/commit/3bb04913239c076c5ead8129cfdda12c72251d5b (simple project)
- Follow setup and run ```cargo run``` once to pull dependencies
- Build again with no changes, and compilation will take 2-4 minutes, always freezing on bootloader

Looking around Cargo issues, it looks like this issue has been reported a few times in the past (https://github.com/rust-lang/cargo/issues/1162, https://github.com/rust-lang/cargo/pull/1714, https://github.com/gimli-rs/cpp_demangle/issues/79), but I didn't find any relevant reported issues. This build time makes running/testing extremely difficult.

BigBadE, 2022-11-26:
Adding onto this:

- Cloning bootloader and depending on it locally doesn't have this issue.
- Using latest bootloader and bootloader_api.

phil-opp, 2022-11-27:
Thanks for reporting! I pushed a fix to #286, which should solve this. Could you try again with the `v0.11.0-beta.4` release?

BigBadE, 2022-11-30:
Confirmed fixed in beta.4

commit 11736be5 (2022-11-26): Remove some `rerun-if-changed` commands to fix #286
PR #283 (2022-12-01): Prepare for v0.11 release and fix remaining issues
Prepares the new `v0.11` merged in #232 for publishing.

TODO:

- [x] update changelog
- [x] create git tags after publishing

Fixes #291 
Fixes #286
closed 2022-11-30 by BigBadE