rust-osdev/bootloader #287: Add cargo features to build only UEFI or BIOS bootloader

phil-opp, 2022-11-27:
Some projects only need one build standard. So we should allow to build only the UEFI or the BIOS bootloader to make it possible to reduce compile times. To achieve this, we could add new `uefi` and `bios` cargo features, which are both enabled by default

Proposed by @AviiNL [on gitter](https://gitter.im/rust-osdev/Lobby?at=63809adcbcdb0060b839bde3).

AlexJMohr, 2022-12-29:
Anyone working on this? I'd like to make an attempt with some guidance.

Obviously `Cargo.toml` needs something like this:
```toml
[features]
default = ["bios", "uefi"]
bios = []
uefi = []
```

And I was thinking I'd add add this to `src/lib.rs`
```rs
#[cfg(feature = "bios")]
mod bios; // src/bios/mod.rs
#[cfg(feature = "uefi")]
mod uefi; // src/uefi/mod.rs
```

phil-opp, 2023-01-02:
Thanks, your approach looks good! The second step is then to update the `build.rs`, where we implement a hacky workaround for building the dependencies (until artifact dependencies are stable in cargo). I think it should be enough to add a `#[cfg(feature = "uefi")] { ... }` block around the `build_uefi_bootloader` call and a `#[cfg(feature = "bios")] { ... }` block around the `build_bios_*` calls in the `main` function. I'm happy to help you with the details, just ping me if you have any questions.

jasoncouture, 2023-01-05:
See jasoncouture/bootloader#1 ( #313 ) and let me know what you think @AlexJMohr

PR #304 (2023-01-09): Add bios and uefi cargo features (closes #287)
closes #287 

- Added `bios` and `uefi` features in `Cargo.toml` (both enabled by default)
- Made `mbrman` and `gpt` dependencies optional, enabled by the `bios` and `uefi` features respectively
- Moved `BiosBoot` to new `src/bios/mod.rs`
- Moved `mbr.rs` to `src/bios` since it is only used in bios boot
- Moved `UefiBoot` to new `src/uefi/mod.rs`
- Moved `gpt.rs` and `pxe.rs` to `src/uefi` since they are only used in uefi boot
- publicly export `BiosBoot` and `UefiBoot` in `lib.rs`, gated by the `bios` and `uefi` features respectively
- Update `bu [... cut]
closed 2023-01-09 by phil-opp