rust-osdev/bootloader #291: .text and .partition_table range overlap on latest nightly

BigBadE, 2022-11-29:
On Windows 10, using Nightly 11/29/2022, the .text and .partition_table range overlap, preventing bootloader from building:
```
    = note: rust-lld: error: section .text file range overlaps with .partition_table
            >>> .text range is [0xC5C, 0xDE1]
            >>> .partition_table range is [0xDBE, 0xDFD]

            rust-lld: error: section .text virtual address range overlaps with .partition_table
            >>> .text range is [0x7C5C, 0x7DE1]
            >>> .partition_table range is [0x7DBE, 0x7DFD]

            rust-lld: error: section .text load address range overlaps with .partition_table
            >>> .text range is [0x7C5C, 0x7DE1]
            >>> .partition_table range is [0x7DBE, 0x7DFD]

  error: could not compile `bootloader-x86_64-bios-boot-sector` due to previous error
```
I pulled commit 2ac7140 to make sure the issue is caused by an update in the toolchain, and previous versions failed to compile as well. The build passed on another Windows 10 pc running a Nightly from earlier this month, I will update on it as well to confirm the issue isn't device-specific.

phil-opp, 2022-11-29:
Thanks for reporting! I think we fixed this error as part of https://github.com/rust-osdev/bootloader/pull/283, which is not merged yet. Could you try the `release` branch? The fixed version is also available on crates.io as ` v0.11.0-beta.4`.

BigBadE, 2022-11-29:
Ah, it was even in with the fundy error, release branch compiles fine. I guess this issue will work for documenting the bug at least.

phil-opp, 2022-11-29:
Thanks for testing! Yeah, this issue is definitely useful until we merge #283, thanks! I pinned it to the top of the issues list.

BigBadE, 2022-11-29:
Thanks! Glad I can help some, still trying to learn more OS dev to directly contribute

PR #283 (2022-12-01): Prepare for v0.11 release and fix remaining issues
Prepares the new `v0.11` merged in #232 for publishing.

TODO:

- [x] update changelog
- [x] create git tags after publishing

Fixes #291 
Fixes #286
closed 2022-12-01 by phil-opp