rust-osdev/bootloader #294: Last page of stack memory is not used

phil-opp, 2022-12-02:
We set the `stack_top` to the start address of the `stack_end` page here: https://github.com/rust-osdev/bootloader/blob/669e3c031f6ee6ad0cbe3f7d09e1b62e8bc34ff1/common/src/lib.rs#L485

However, it looks like `stack_end` is an inclusive bound: https://github.com/rust-osdev/bootloader/blob/669e3c031f6ee6ad0cbe3f7d09e1b62e8bc34ff1/common/src/lib.rs#L183-L186

So I think the stack pointer should be initialized with `(stack_end + 1).start_address()` instead.

Reported by `klim` on the Discord of [Theseus OS](https://www.theseus-os.com/).

inhibitor1217, 2023-01-27:
It would be a lot of help if this issue is fixed soon. 

I'm currently working with a setup which assigns a single page (4 KiB) for the stack of each thread, including the kernel thread. Currently, last page (which is, the only page) of the stack is not used, causing the page fault to occur immediately after we jump to the kernel.

phil-opp, 2023-01-27:
@inhibitor1217 Thanks for the reminder! I'll prepare a PR.

PR #335 (2023-02-06): Create kernel stack with correct size and set up a guard page
The last page of stack memory was not used before (see #294 ) because we initialized the stack pointer with the start address of the inclusive end page of the stack. This PR fixes this by initializing the stack pointer with the exact configured address, aligned to a 16-byte boundary. To verify that this works, I added a new `min_stack` test that tries booting with a kernel stack size of just 3000 bytes.

We also claimed in the docs that we set up a guard page for the kernel stack, but never actually did this. This PR fixes that issu [... cut]
closed 2023-02-06 by phil-opp