rust-osdev/bootloader #306: Better logging

jasoncouture, 2023-01-03:
The UEFI and bios loggers use only the framebuffer.

This can be an issue during headless runs (server with no display, unit tests)

I propose we refactor the logger to support multiple methods of logging output. (Serial being the first idea), with support in the kernel configuration to enable/disable sources. Or a text config file (this is likely better, so it can be configured at runtime)

This would also allow us to separate the logging and framebuffer init, as well be able to add/remove loggers at will. Rather than using `writeln` for early logging, we will be able to initialize the logger before anything else, and configure it with the optional `config.txt` or a default config that matches the current behavior.

phil-opp, 2023-01-03:
Good idea! I fully agree that optional serial logging would be useful.

> Or a text config file (this is likely better, so it can be configured at runtime)

You mean a separate text file that is placed on the FAT partition? I guess we could create such a file in the `create_disk_image` function. The question is how we split the options into kernel and runtime options. Candidates for runtime options could be the recently merged #303 and also the framebuffer size requirements.

phil-opp, 2023-01-04:
Sounds reasonable to me! How about we start with a kernel config option for configuring the logger and then do the runtime config file separately?

asensio-project, 2023-01-04:
I think that the most of the crates that can be helpful for interpreting a configuration file depend on alloc. And if we create our own crate, we could want allocation.

So, what about implementing first the allocation in the bootloader.

asensio-project, 2023-01-04:
I am working in this and I think that we should move the code of the framebuffer to ``common/src/framebuffer.rs`` and the code related to the logger to ``common/src/logger.rs``.

This will be alright for then adding the serial being and modifying the code of the logger for including this method.

Do you like this aproach or do you have another idea?

phil-opp, 2023-01-05:
Thanks for tackling this! 

> I think that we should move the code of the framebuffer to `common/src/framebuffer.rs` and the code related to the logger to `common/src/logger.rs`.
> 
> This will be alright for then adding the serial being and modifying the code of the logger for including this method.

Sounds good!

PR #314 (2023-01-09): Improve Logging
Hello,

This PR improves the actual logging of the bootloader.

Tasks:

- [x] Separate the logger from the framebuffer
- [x] Add support for Serial Being logger
- [x] Add configurations for enabling or disabling these methods of logging

Closes #306

Thanks!