rust-osdev/bootloader #315: Frame allocator assumes memory map is in ascending start address order

tsoutsman, 2023-01-05:
Currently, `LegacyFrameAllocator::allocate_frame_from_descriptor` only ever increases `self.next_frame`. This is problematic, as the memory map descriptors aren't guaranteed to be in ascending start address order (at least for UEFI). If the first descriptor provided by the memory map were also the one with the largest start address, the frame allocator would not allocate from the other descriptors.

Rather than checking whether `self.next_frame < self.start_frame` in `allocate_frame_from_descriptor`, we should set `self.next_frame` to the start address of the descriptor when we find the next suitable descriptor in `allocate_frame`.

phil-opp, 2023-01-05:
Good catch, thanks a lot for reporting! Your proposed solution sounds good to me. To summarize:

- Add a new `min_frame` field to `LegacyFrameAllocator` and set this field in `new_starting_at`. The `next_frame` field should be initialized to 0.
- Set `self.next_frame` to `core::cmp::max(descriptor.start(), self.min_frame)` when selecting a new descriptor. The assignment should happen just after the `Usable` check here: https://github.com/rust-osdev/bootloader/blob/fffa7ceb317328ed19bc0ebaa0028be152095a05/common/src/legacy_memory_region.rs#L250-L254
- Assert that `self.next_frame >= start_frame` in `allocate_frame_from_descriptor` (since this should always be the case now) instead of the current comparison: https://github.com/rust-osdev/bootloader/blob/fffa7ceb317328ed19bc0ebaa0028be152095a05/common/src/legacy_memory_region.rs#L66-L69

I'd be happy to merge a PR that implements the above.

tsoutsman, 2023-01-05:
Looking at the code for `construct_memory_map`, that would need to be changed as well. We would need to compare `original` and `memory_map` to see which memory regions have already been consumed by the allocator.

phil-opp, 2023-01-05:
Oh yes, good point!

Freax13, 2023-02-16:
Could we just sort the memory map?

JarlEvanson, 2023-02-16:
I don't think Iterators are sortable, so we would need to be able to collect the iterator, which would have to be able to be done on the stack, since we don't have access to a heap or dynamic allocation other than that. I don't think it's possible

Freax13, 2023-02-16:
For BIOS the iterator is created from a slice. If we sort the slice, the iterator is also sorted. For UEFI the iterator is not created from a Rust slice but some raw bytes that are casted to descriptors. The uefi crate currently doesn't have an API to sort those descriptors, but I think it might be possible to add that.

JarlEvanson, 2023-02-17:
I am working on getting sorting implemented in the uefi crate, and then will start work on the bios sorting.

Freax13, 2023-02-17:
Bios is already sorted:
https://github.com/rust-osdev/bootloader/blob/52a0a679bf7190538e96d211c96f41489e04a935/bios/stage-4/src/main.rs#L35

phil-opp, 2023-04-09:
Once #360 is merged, we will be able to sort the UEFI memory map as well using the new [`MemoryMap::sort`](https://docs.rs/uefi/0.20.0/uefi/table/boot/struct.MemoryMap.html#method.sort) method.

phil-opp, 2023-04-13:
I just merged #360. I PR to sort the memory map would be appreciated!

closed 2023-04-17 by phil-opp