rust-osdev/bootloader #327: bootloader v0.11 broken under bochs and qemu/kvm

Stary2001, 2023-01-15:
Two issues here: 
* `enter_unreal_mode` should be the very first thing in stage2 main, because the screen write call touches some memory above 64k (`movzx ebx, byte ptr ds:[ecx]` with `ecx=0x13ccc`)
error output in bochs: `00017458470e[CPU0  ] read_virtual_checks(): read beyond limit`

* Rust generates some code that uses `ss` for accesses (`mov eax, dword ptr ss:[ebp+edi*8+4]`, where ss is the default for ebp), and enter_unreal_mode only sets up `ds`. Unsure how this should be fixed properly - set ss to 32 bit as well?
error output in bochs: `00695082304e[CPU0  ] stackPrefetch(4): access [0x00013ff8] > SS.limit [0x0000ffff]`

As for why this works in qemu in emulation mode: qemu is really inaccurate :(

phil-opp, 2023-01-16:
Thanks a lot for reporting these issues! Your proposed fixes sound good.

Stary2001, 2023-01-16:
Do you want me to make a PR?

phil-opp, 2023-01-16:
That would be great!

PR #328 (2023-01-16): Correctness fixes for stage2
Correctness fixes to allow the bootloader to work on bochs and kvm.
closes #327
closed 2023-01-16 by phil-opp