rust-osdev/bootloader #329: thread 'main' panicked at 'no entry found for key', src\tools\cargo\src/cargo\core\compiler\unit_dependencies.rs

LovesAsuna, 2023-01-20:
As you submitted the [issue](https://github.com/rust-lang/cargo/issues/10444)earlier，I don't know how to solve this problem.

phil-opp, 2023-01-20:
Could you clarify what you're trying to do and how this is related to the `bootloader` crate?

LovesAsuna, 2023-01-20:
I am now migrating from bootloader v0.9.X to bootloader v0.11. I follow the guide and  use the build script to build kernel.But ran into the problem I mentioned above.

---

In the root package, I created a `config.toml`

```toml
[unstable]
bindeps = true
build-std-features = ["compiler-builtins-mem"]
build-std = ["core", "compiler_builtins", "alloc"]

[build]
target = "x86_64-simple_os.json"
```

In the root package, the `Cargo.toml` is

```toml
[package]
name = "simple_os"
version = "0.1.0"
edition = "2021"

[build-dependencies]
bootloader = "0.11"
kernel = { path = "kernel", artifact = "bin", target = "x86_64-unknown-none"}

[workspace]
members = ["kernel"]
```

In the kernel package, I created a `config.toml` as well

```toml
[unstable]
build-std-features = ["compiler-builtins-mem"]
build-std = ["core", "compiler_builtins", "alloc"]

[build]
target = "x86_64-simple_os.json"
```

The problem is that when I execute command `Cargo build`, the error occurs.

[image]

---

By the way, I've tried leaving only `bindeps` option in the root package's `config.toml`.Then the error would be

[image]

phil-opp, 2023-01-20:
Thanks for the information! You don't need the custom target file and build-std config keep keys anymore.

> By the way, I've tried leaving only bindeps option in the root package's config.toml.Then the error would be

That's the right approach. To fix that error, run `rustup target add x86_64-unknown-none`.

LovesAsuna, 2023-01-20:
I've added target `x86_64-unknown-none`. But the error is still not resolved
[image]

phil-opp, 2023-01-20:
Try removing the build.target key from your Cargo.toml. And make sure that your kernel is declared as no_std.

LovesAsuna, 2023-01-21:
Thanks, now I can successfully build and run the os. But I have another problem that with BIOSbooting, the println macro doesn't seem to work while with UEFI booting, it does work

closed 2023-01-21 by LovesAsuna

phil-opp, 2023-01-29:
> But I have another problem that with BIOSbooting, the println macro doesn't seem to work while with UEFI booting, it does work.

What does your `println` macro do? Be aware that the `v0.11` version of the bootloader sets up a pixel-based framebuffer, so the VGA text mode won't work anymore (it's not compatible with UEFI anyway). See https://github.com/rust-osdev/bootloader/blob/main/docs/migration/v0.9.md for details.

phil-opp, 2023-01-30:
We also set up a framebuffer for BIOS booting so that you don't need to special-case your output code depending on the boot mode.

(The screenshot you posted above just shows the output from the bootloader, no kernel output yet.)

LovesAsuna, 2023-01-30:
My output is based on framebuffer and doesn't change anything else. Completely consistent code applied to UEFI booting is correct, while BIOS booting is wrong