rust-osdev/bootloader #331: Panic in stage 4 in BIOS boot

an-owl, 2023-01-21:
The bootloader panics on line 242 with `failed to identity map fame PhysFrame[4kib](20a000): PageAlreadyMapped(PhysFrame[4Kib](0x20a000))`

https://github.com/rust-osdev/bootloader/blob/c73786d894c4f32c12835c4ed1c02b412e3a668d/common/src/lib.rs#L229-L244

I checked the mapping for `0x20b000` and QEMU says its mapped to `0x20b000`

A potental fix would be to just ignore the error with an extra match arm ` Err(MapToError::PageAlreadyMapped(frame)) => {} `

phil-opp, 2023-01-21:
Thanks for reporting! Could you give us instructions how to reproduce this error?

Freax13, 2023-01-21:
I suspect that this could be caused by a non-PIE kernel that maps memory to the same address we use for the context switch in the bootloader. The error occurs because we load the context switch function into the kernel's address space. We currently need to do this because the address space switch is not the last instruction of the context switch, so the context switch function needs to be mapped in the kernel's address space to complete.

Similar errors could probably be provoked by setting other addresses (e.g. frame buffer, boot info) to this address.

This might be fixed by #240.

> A potental fix would be to just ignore the error with an extra match arm `Err(MapToError::PageAlreadyMapped(frame)) => {}`

No, this wouldn't work, the current implementation expects the context switch function to be mapped in both address spaces. Ignoring the error would mean that the context switch function would jump into rogue kernel memory after doing the address space switch.

an-owl, 2023-01-21:
After removing `"-C","relocation-model=static"` and `"-C","link-arg=-no-pie"` from rustflags the bootloader now boots correctly so it seems that is the likely cause.

> No, this wouldn't work, the current implementation expects the context switch function to be mapped in both address spaces. Ignoring the error would mean that the context switch function would jump [...]

I was supposed to put a match guard in there. Using `gva2gpa` on the page that threw the error and the ones before and after it all seem to be identity mapped already.

phil-opp, 2023-01-26:
> After removing `"-C","relocation-model=static"` and `"-C","link-arg=-no-pie"` from rustflags the bootloader now boots correctly so it seems that is the likely cause.

I just tried to reproduce the error by adding these flags, but it still works fine for a small test kernel. Do you have the project where this happens online somewhere so that we can try to reproduce the error?

an-owl, 2023-01-27:
> Do you have the project where this happens online somewhere so that we can try to reproduce the error?

Yep its [here](https://github.com/an-owl/hootux)

phil-opp, 2023-01-27:
Thanks a lot! I created a PR to avoid that issue for now at https://github.com/rust-osdev/bootloader/pull/337.

In the long term, I agree that we need a more general fix for this, such as #240 (which I still haven't had time to review yet unfortunately).

PR #337 (2023-01-29): Map BIOS stage-4 at lower address to avoid conflicts with the kernel
Parts of stage-4 need to be identity-mapped. This can lead to 'page already mapped' errors when the kernel already occupies the virtual address corresponding to the physical address of the stage-4. For example, a typical virtual start address of the kernel is 2MiB. This was where we placed the stage-4 in physical memory before. This commit avoids that conflict by keeping the stage-4 below the 2MiB boundary.

Fixes #331
closed 2023-01-29 by phil-opp