rust-osdev/bootloader #338: Terminal blocked after cargo test

asensio-project, 2023-01-27:
Hello,

After I execute ``cargo test`` the STDOUT of my terminal blocks, so I cannot see what I am writing. This happens after #333. I think that it's an unclosed pipe  in the run_qemu() function, but I don't know anything more.

I am working for fixing it. And I will publish a PR. But if someone knows something...

Thanks!

phil-opp, 2023-01-27:
You mean while `cargo test` runs? Or still afterwards?

In case you mean the former: that's probably because we're forwarding stdin to QEMU. Let me try to fix this.

phil-opp, 2023-01-27:
I opened https://github.com/rust-osdev/bootloader/pull/339 to set `stdin` to `null` for the QEMU command. Could you try whether this fixes your issue?

asensio-project, 2023-01-28:
I tried it, and it fixes it. Thanks!

PR #339 (2023-01-29): Test framework: Don't inherit `stdin` when spawning QEMU
Fixes #338
closed 2023-01-29 by phil-opp