rust-osdev/bootloader #348: A simple hello world does not display in Qemu

piosystems, 2023-03-12:
I followed the instructions on [https://github.com/rust-osdev/bootloader/blob/main/README.md](https://github.com/rust-osdev/bootloader/blob/main/README.md) and  [https://github.com/rust-osdev/bootloader/blob/main/docs/create-disk-image.md](https://github.com/rust-osdev/bootloader/blob/main/docs/create-disk-image.md) and got all successfully setup

A simple code  hello world code like that shown below in my test kernel, builds but when I run, Qemu opens quite alright and shows a series of info. Unfortunately, it never ends up showing my Hello World. I am not using any custom linker. Am I missing something?

`#![no_std]`
#![no_main]
#![feature(alloc_error_handler)]

bootloader_api::entry_point!(my_entry_point);

fn my_entry_point(bootinfo: &'static mut bootloader_api::BootInfo) -> ! {
 let hello = b"Hello World";
    for (i, &byte) in hello.iter().enumerate() {
        unsafe {
            *display_cursor_position.offset(i as isize * 2) = byte;
            *display_cursor_position.offset(i as isize * 2 + 1) = 0x0f;
        }
    }
`loop {}
}

#[panic_handler]
fn panic(_info: &core::panic::PanicInfo) -> ! {
    loop {}
}

/// The global allocator type.
#[derive(Default)]
pub struct Allocator;

unsafe impl GlobalAlloc for Allocator {
    unsafe fn alloc(&self, layout: Layout) -> *mut u8 {
        alloc(layout) as *mut u8
    }
    unsafe fn dealloc(&self, ptr: *mut u8, _layout: Layout) {
        dealloc(ptr, _layout);
    }
}

/// If there is an out of memory error, just panic.

#[alloc_error_handler]
fn my_allocator_error(_layout: Layout) -> ! {
    panic!("out of memory");
}

/// The static global allocator.
#[global_allocator]
`static GLOBAL_ALLOCATOR: Allocator = Allocator;`

phil-opp, 2023-03-12:
We don't use the VGA text mode because it is not compatible with UEFI booting. Instead, we set up a pixel-based framebuffer, which you can access through the [`bootinfo.framebuffer`](https://docs.rs/bootloader_api/0.11.0/bootloader_api/info/struct.BootInfo.html#structfield.framebuffer) field.

As this buffer is pixel-based instead of text-based, you have to do the font rendering yourself. I recommend to use the [`noto_sans_mono_bitmap`](https://docs.rs/noto-sans-mono-bitmap/latest/noto_sans_mono_bitmap/index.html) for that as a start. See our internal [`framebuffer`](https://github.com/rust-osdev/bootloader/blob/main/common/src/framebuffer.rs) submodule as an example.

Hope this helps!

piosystems, 2023-03-13:
> We don't use the VGA text mode because it is not compatible with UEFI booting. Instead, we set up a pixel-based framebuffer, which you can access through the [...]

This approach worked. I was able to see the "Hello, World". Thank you.