rust-osdev/bootloader #352: Support for global allocator

piosystems, 2023-03-14:
Is there any support for global allocator. Concatenating string slices is not feasible without heap allocator.

piosystems, 2023-03-15:
I have attempted using the following code to setup default global allocator. Any concrete example on how this could work with bootloader?

use alloc::alloc::*;

/// The global allocator type.
#[derive(Default)]
pub struct Allocator;

unsafe impl GlobalAlloc for Allocator {
     unsafe fn alloc(&self, layout: Layout) -> *mut u8 {
         malloc(layout.size() as u32) as *mut u8
     }
     unsafe fn dealloc(&self, ptr: *mut u8, _layout: Layout) {
         free(ptr as *mut c_void);
     }
}

/// If there is an out of memory error, just panic.
#[alloc_error_handler]
fn my_allocator_error(_layout: Layout) -> ! {
    panic!("out of memory");
}

/// The static global allocator.
#[global_allocator]
static GLOBAL_ALLOCATOR: Allocator = Allocator;

phil-opp, 2023-03-17:
The bootloader does no longer exist after transferring control to your kernel. So you have to implement the memory management yourself. This includes page table table allocation and creating your own allocator. See the "Memory Management" section on <https://os.phil-opp.com/> for details.

If you don't want to manage memory, you could create an UEFI application directly. See <https://github.com/rust-osdev/uefi-rs/tree/main/template> for an example. By using the `uefi` crate you have direct access to the UEFI boot services provided by the firmware. This allows you to allocate pages through the [`BootServices::allocate_pool`](https://docs.rs/uefi/0.19.1/uefi/table/boot/struct.BootServices.html#method.allocate_pool) function. You can also include the higher-level [`uefi-services`](https://docs.rs/uefi-services/0.16.0/uefi_services/) crate, which will automaticall set up a global allocator for you.

Hope this help!

piosystems, 2023-03-21:
> Thank you for the feedback. I am following your "Memory Management" section on https://os.phil-opp.com/ but the flow seem not to be compatible with bootloader_api 0.11.x. For example, the newer bootinfo has no memory_map member anymore which is expected by memory::BootInfoFrameAllocator. Instead, there is memory_regions which is available for use. Can you post a quick example update of memory allocation that works with bootloader_api 0.11.x?

I am still stuck with the fact that for `bootloader_api 0.11.x`, the `boot_info: &'static mut bootloader_api::BootInfo` passed to my `entry_point`  does not have `memory_map`, contrary to what is indicated in [https://os.phil-opp.com/](https://os.phil-opp.com/) which seems to have been written for earlier versions. Attempts to find ways to use the Allocators in [https://os.phil-opp.com/](https://os.phil-opp.com/) have been unsuccessful. Am I missing something? Can you give an example of the use of FixedSizeBlockAllocator with `bootloader_api 0.11.x`?

piosystems, 2023-03-22:
> 

> I'm pretty sure its been replaced with `memory_regions`

How do you initialize any Allocator of choice with addresses indicated in memory_regions? Any working example?

yavko, 2023-03-22:
> > 
> 
> > I'm pretty sure its been replaced with `memory_regions`
> 
> How do you initialize any Allocator of choice with addresses indicated in memory_regions? Any working example?

i think like so, though havent been able to test:
## initialization
[image]

## Updated frame allocator
[image]
[image]

piosystems, 2023-03-22:
> > > I'm pretty sure its been replaced with `memory_regions`
> > 
> > 
> > How do you initialize any Allocator of choice with addresses indicated in memory_regions? Any working example?
> 
> i think like so, though havent been able to test (with `BootInfoFrameAllocator`):
> 
> ## initialization
> [image]
> 
> ## Updated frame allocator
> [image] [image] [image]

From what I can see in your posted snippets, it is obvious that may codes are not up to date. Can you point me to the updated repository content url.