rust-osdev/bootloader #353: target x86_64-unknown-none dependancy conflict

Firas88Alkhatib, 2023-03-16:
when I run : "cargo build --target x86_64-unknown-none"

I get the error :

268 | /         compile_error!("target is not supported, for more information see: \
269 | |                         https://docs.rs/getrandom/#unsupported-targets");
    | |________________________________________________________________________^

phil-opp, 2023-03-16:
Do you have your full code online somewhere so that I can take a look? It's difficult to understand the problem with just the error output.

My guess is that you're trying to compile the `bootloader` crate for the `x86_64-unknown-none` target, which is not how it should be used. Instead, your kernel should depend on the **`bootloader_api`** crate (which can be compiled for `x86_64-unknown-none`) and the `bootloader` crate should be built for your host system (without any `--target` argument).

bjorn3, 2023-03-16:
In which directory are you running that command? It looks like you are running it in the directory of a crate that depends on getrandom, which indeed doesn't support the absence of an OS.

Firas88Alkhatib, 2023-03-16:
I was trying to run a very basic kernel in qemu to just get started.
so in I've created a new Rust project and I added this code to main.rs:
```
#![no_std]
#![no_main]

fn kernel_main(boot_info: &'static mut bootloader_api::BootInfo) -> ! {
    loop {}
}

bootloader_api::entry_point!(kernel_main);
```

and then in Cargo.toml:
```
[package]
name = "something"
version = "0.1.0"
edition = "2021"

[dependencies]
bootloader = "0.11.2"
bootloader_api = "0.11.2"

```
and then "cargo build --target x86_64-unknown-none" from the same folder. no subfolders , nothing else.

now I've noticed that its not how it should be.

Do you think it is a good idea to provide a very basic example in a folder called examples in this repo to help people like me to get started ?
note: the common folder is too much to get started as a lot of useful stuff are there that may confuse starters.

by the way the tutorial by Philipp Oppermann "https://os.phil-opp.com/" is extremely useful, thank you @phil-opp and all contributors involved to make this happen!.

phil-opp, 2023-03-16:
Ah yes, you should only include the `bootloader_api` dependency for your kernel, not the `bootloader` crate itself. Also, you need to use the `entry_point` macro to make your kernel bootable.

> Do you think it is a good idea to provide a very basic example in a folder called examples in this repo to help people like me to get started ?

Yes, we should definitely do that.

> by the way the tutorial by Philipp Oppermann "https://os.phil-opp.com/" is extremely useful, thank you @phil-opp and all contributors involved to make this happen!.

Great to hear, thanks a lot :). I'll try my best to update it to the new bootloader version as soon as possible.

closed 2023-03-16 by Firas88Alkhatib

yavko, 2023-03-22:
In my kernel crate, I'm only loading `bootloader_api`, however it still happens to me. Here is my repo https://github.com/yavko/yavkOS/tree/update-bootloader

bjorn3, 2023-03-22:
It looks like something enables the std feature of futures-core.

yavko, 2023-03-25:
> It looks like something enables the std feature of futures-core.

That should be the case though, since `futures-core` is only required in the top level crate not kernel crate, in kernel crate these dependencies don't exist.
[image]

phil-opp, 2023-03-25:
Thanks a lot for reporting! I checked out your code and it looked all fine, but I could reproduce your error. It took me quite a bit of trial and error until I minimized your code enough to find the issue. It turns out that you're not specifying an `edition` or `resolver` in your top-level `Cargo.toml`, which causes cargo to use the old feature resolver. As a results, it unifies dependencies across targets, which turns the std-dependent features on for your kernel dependencies as well. See https://blog.rust-lang.org/2021/03/25/Rust-1.51.0.html#cargos-new-feature-resolver for more information.

I opened a PR on your repo to fix this by setting the `edition` to `2021`, where the new resolver is the default: https://github.com/yavko/yavkOS/pull/1

yavko, 2023-03-25:
> Thanks a lot for reporting! I checked out your code and it looked all fine, but I could reproduce your error. It took me quite a bit of trial and error until I minimized your code enough to find the issue. It turns out that you're not specifying an `edition` or `resolver` in your top-level `Cargo.toml`, which causes cargo to use the old feature resolver. As a results, it unifies dependencies across targets, which turns the std-dependent features on for your kernel dependencies as well. See [blog.rust-lang.org/2021/03/25/Rust-1.51.0.html#cargos-new-feature-resolver](https://blog.rust-lang.org/2021/03/25/Rust-1.51.0.html#cargos-new-feature-resolver) for more information.
> 
> I opened a PR on your repo to fix this by setting the `edition` to `2021`, where the new resolver is the default: [yavko/yavkOS#1](https://github.com/yavko/yavkOS/pull/1)

Thank you so much for finding this!