rust-osdev/bootloader #354: BIOS stage 2 subtract overflows

hawkw, 2023-03-16:
I'm attempting to migrate my kernel from v0.10.x of `bootloader`, and after switching to v0.11.x and running in QEMU, I see a panic in the BIOS bootloader's stage 2 code:
[image]

The subtract that overflows appears to be on this line: https://github.com/rust-osdev/bootloader/blob/fa87d01a4a166b51412d162c8343bc2565bdbee0/bios/stage-2/src/fat.rs#L96-L99

I'm creating the image using `BiosBoot::create_disk_image`, and passing it to QEMU with the arguments `-drive format=raw,file=/path/to/bios.img` as suggested in the [documentation on creating a disk image](https://github.com/rust-osdev/bootloader/blob/main/docs/create-disk-image.md), so I don't _think_ anything about my configuration is particularly out of the ordinary. I can provide additional information if there's anything that would help, though.

**Bootloader version:** 0.11.2
**QEMU version**:
```console
$ qemu-system-x86_64 --version
QEMU emulator version 7.2.0
Copyright (c) 2003-2022 Fabrice Bellard and the QEMU Project developers
```

phil-opp, 2023-03-17:
Thanks for reporting! I'm not sure what's going wrong here yet. Do you have your code online somewhere so that I can do some debugging?

hawkw, 2023-03-17:
@phil-opp The branch updating my OS to use `bootloader` v0.11.2 is here: https://github.com/hawkw/mycelium/pull/406

Unfortunately, it's definitely not a minimal reproduction of the crash. I did try running the tests in the `bootloader` repo on my machine with the same version of QEMU, and they all passed for me. So, it seems like *something* about my code is required to reproduce this...I'm just not sure what. Happy to keep debugging if there's anything I can do to help.