rust-osdev/bootloader #361: ovmf-prebuilt v0.1.0-alpha.1 doesn't compile

Zingoooooo, 2023-03-29:
The result i get when I compile with cargo 1.70.0-nightly (15d090969 2023-03-21) is listed below. What am I missing?
Im fillowing the migration guide from 0.9.

 Compiling ovmf-prebuilt v0.1.0-alpha.1
error[E0463]: can't find crate for `std`
  |
  = note: the `x86_64-unknown-none` target may not support the standard library
  = note: `std` is required by `ovmf_prebuilt` because it does not declare `#![no_std]`
  = help: consider building the standard library from source with `cargo build -Zbuild-std`

error[E0463]: can't find crate for `std`
 --> /home/zingo/.cargo/registry/src/index.crates.io-6f17d22bba15001f/ovmf-prebuilt-0.1.0-alpha.1/src/lib.rs:1:5
  |
1 | use std::path::{PathBuf, Path};
  |     ^^^ can't find crate
  |
  = note: the `x86_64-unknown-none` target may not support the standard library
  = help: consider building the standard library from source with `cargo build -Zbuild-std`

error: cannot find macro `assert` in this scope
 --> /home/zingo/.cargo/registry/src/index.crates.io-6f17d22bba15001f/ovmf-prebuilt-0.1.0-alpha.1/src/lib.rs:5:5
  |
5 |     assert!(path.exists(), "OVMF-pure-efi.fd does not exist");
  |     ^^^^^^
  |
  = help: consider importing this macro:
          core::assert

error: cannot find macro `env` in this scope
 --> /home/zingo/.cargo/registry/src/index.crates.io-6f17d22bba15001f/ovmf-prebuilt-0.1.0-alpha.1/src/lib.rs:4:26
  |
4 |     let path = Path::new(env!("CARGO_MANIFEST_DIR")).join("OVMF-pure-efi.fd");
  |                          ^^^
  |
  = help: consider importing this macro:
          core::env

phil-opp, 2023-10-25:
Could you give more details about the commands that you're running and your project setup? Or do you have your project online somewhere so that I can take a look?

In general, the `ovmf-prebuilt` should be built for your host system, not for the `x86_64-unknown-none` target.

br0kenpixel, 2024-04-23:
I ran into the same issue. Turns out, my `.cargo/config.toml` was also forcing `ovmf-prebuilt` to be built for the `x86_64-unknown-none` target. I fixed it by moving the `[build]` section to a separate `.cargo/config.toml` inside my kernel crate.

```
.
├── .cargo
│   └── config.toml <-- only contains `[unstable] bindeps = true`
├── Cargo.lock
├── Cargo.toml
├── build.rs
├── kernel
│   ├── .cargo
│   │   └── config.toml <-- only contains `[build] target = "x86_64-unknown-none"`
│   ├── Cargo.toml
│   └── src
├── rust-toolchain.toml
├── src
│   └── main.rs
└── target
```