rust-osdev/bootloader #370: bootloader not compiling due to missing `compiler-builtins-mem` with latest nightly

iTitus, 2023-05-05:
After the latest nightly update the bootloader 0.11.3 dependency does not compile anymore.
Probably caused by https://github.com/rust-lang/rust/pull/108865.

```
cargo run
   Compiling bootloader v0.11.3
error: failed to run custom build command for `bootloader v0.11.3`                                                                                                      
                                                                                                                                                                        
Caused by:                                                                                                                                                              
  process didn't exit successfully: `C:\Users\iTitus\Documents\Coding\HHU\BSE\hhu_os\target\debug\build\bootloader-3f3360a7d583746b\build-script-build` (exit code: 101)
  --- stderr                                                                                                                                                            
      Updating crates.io index                                                                                                                                          
      Blocking waiting for file lock on package cache                                                                                                                   
      Blocking waiting for file lock on package cache                                                                                                                   
      Blocking waiting for file lock on package cache                                                                                                                   
      Updating crates.io index                                                                                                                                          
    Installing bootloader-x86_64-bios-boot-sector v0.11.3                                                                                                               
      Blocking waiting for file lock on package cache                                                                                                                   
      Updating crates.io index
    Installing bootloader-x86_64-bios-stage-2 v0.11.3
      Blocking waiting for file lock on package cache
      Updating crates.io index
    Installing bootloader-x86_64-bios-stage-4 v0.11.3
      Blocking waiting for file lock on package cache
      Blocking waiting for file lock on package cache
      Updating crates.io index
    Installing bootloader-x86_64-bios-stage-3 v0.11.3
      Blocking waiting for file lock on package cache
      Blocking waiting for file lock on package cache
      Updating crates.io index
      Blocking waiting for file lock on package cache
      Updating crates.io index
      Updating crates.io index
      Blocking waiting for file lock on package cache
      Blocking waiting for file lock on package cache
      Blocking waiting for file lock on package cache
  warning: package `spin v0.9.5` in Cargo.lock is yanked in registry `crates-io`, consider running without --locked
      Blocking waiting for file lock on package cache
      Updating crates.io index
      Blocking waiting for file lock on package cache
      Blocking waiting for file lock on package cache
      Blocking waiting for file lock on package cache
      Updating crates.io index
      Updating crates.io index
      Blocking waiting for file lock on package cache
      Blocking waiting for file lock on package cache
      Updating crates.io index
      Blocking waiting for file lock on package cache
  error: failed to compile `bootloader-x86_64-bios-boot-sector v0.11.3`, intermediate artifacts can be found at `C:\Users\iTitus\AppData\Local\Temp\cargo-install096Gcd`

  Caused by:
    Package `test v0.0.0 (C:\Users\iTitus\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\test)` does not have the feature `compiler-builtins-mem`
  thread 'main' panicked at 'failed to build bios boot sector', C:\Users\iTitus\.cargo\registry\src\index.crates.io-6f17d22bba15001f\bootloader-0.11.3\build.rs:151:9
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
      Blocking waiting for file lock on package cache
      Blocking waiting for file lock on package cache
  error: failed to compile `bootloader-x86_64-bios-stage-2 v0.11.3`, intermediate artifacts can be found at `C:\Users\iTitus\AppData\Local\Temp\cargo-installFtwmto`
  error: failed to compile `bootloader-x86_64-bios-stage-4 v0.11.3`, intermediate artifacts can be found at `C:\Users\iTitus\AppData\Local\Temp\cargo-installSRvl3I`

  Caused by:
    Package `test v0.0.0 (C:\Users\iTitus\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\test)` does not have the feature `compiler-builtins-mem`
  error: failed to compile `bootloader-x86_64-bios-stage-3 v0.11.3`, intermediate artifacts can be found at `C:\Users\iTitus\AppData\Local\Temp\cargo-installmnRzgw`

  Caused by:
    Package `test v0.0.0 (C:\Users\iTitus\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\test)` does not have the feature `compiler-builtins-mem`
```

bjorn3, 2023-05-05:
`-Zbuild-std` is broken with latest nightly due to a change of the standard library: https://rust-lang.zulipchat.com/#narrow/stream/246057-t-cargo/topic/build-std.20failing

deathly809, 2023-05-05:
Thanks @bjorn3. I was just looking at the bootloader repo and got the same issue there. So I was about to close this :)

iTitus, 2023-05-20:
was fixed in rust

closed 2023-05-20 by iTitus