rust-osdev/bootloader #375: Cannot compile version 0.11.3

DecrepitHuman, 2023-05-28:
I've followed the quick start provided in this repo (https://github.com/rust-osdev/bootloader/blob/HEAD/docs/create-disk-image.md)
I've got the exact dependencies, files, directories, layout and so on. I'm compiling to `x86_64-unknown-none` with rustc version 1.71.0-nightly in WSL2 (Ubuntu)

I get the following output when specifically creating a UEFI image: https://sourceb.in/2wDVzkJ4ZV
I have LLVM15 installed and in my PATH (`llvm-config --version` shows 15.0.7)

bjorn3, 2023-05-28:
Have you installed the `llvm-tools` rustup component using `rustup component add llvm-tools`? Bootloader's build script looks for the objcopy llvm tool exclusively in the rustc toolchain, not in your `PATH`.

DecrepitHuman, 2023-05-28:
That fixed it. I can't recall reading anywhere that I needed to do this, but I know for a fact that I wasn't reading everything. Thank you, it's now compiling

closed 2023-05-28 by DecrepitHuman