rust-osdev/bootloader #376: Struggling to get function kernel running

DecrepitHuman, 2023-05-29:
This is my first project using this crate, I've followed the quick start (https://github.com/rust-osdev/bootloader/blob/HEAD/docs/create-disk-image.md) exactly down to the file/folder names. The project generates both `uefi.img` and `bios.img` files which QEMU accepts, but after loading the kernel I can see "Third Stage (....)" in the console, but it immediately closes and restarts the kernel. The entry point (using the `entry_point!()` macro) has a infinite loop and so does the panic handler, which I would assume would leave the kernel "hanging" but instead constantly restarts as if the kernel's code isn't running at all.

asensio-project, 2023-06-12:
That document doesn't explains how to add something useful to the kernel. Have you really got a kernel?
If you don't have one, the bootloader will automatically enter in panic. And QEMU will handle it rebooting the system.

phil-opp, 2023-06-25:
I'm currently working on a more detailed guide for the upcoming third edition of os.phil-opp.com. The latest draft is available at https://github.com/phil-opp/blog_os/blob/edition-3/blog/content/edition-3/posts/02-booting/index.md. Could you try whether this fixes your issues?