rust-osdev/bootloader #382: bootloader_api Bootloader_Config aslr causes crash

interstellarfrog, 2023-07-04:
when using aslr it causes a crash before even entering the kernel is there any specific requirements to use this?

Bootloader_Config:
```
pub static BOOTLOADER_CONFIG: BootloaderConfig = {
    let mut config = BootloaderConfig::new_default();
    config.mappings.physical_memory = Some(Mapping::Dynamic);
    config.mappings.page_table_recursive = None;
    config.mappings.aslr = true;
    config
};

entry_point!(kernel_main, config = &BOOTLOADER_CONFIG);
```

Log: 

```
INFO : Framebuffer info: FrameBufferInfo { byte_len: 2764800, width: 1280, height: 720, pixel_format: Bgr, bytes_per_pixel: 3, stride: 1280 }

INFO : 4th Stage

INFO : BiosInfo { stage_4: Region { start: 130000, len: 2b2f8 }, kernel: Region { start: 1000000, len: 223ad0 }, ramdisk: Region { start: 1224000, len: 104 }, config_file: Region { start: 1224104, len: d2 }, last_used_addr: 12241d5, framebuffer: BiosFramebufferInfo { region: Region { start: fd000000, len: 2a3000 }, width: 500, height: 2d0, bytes_per_pixel: 3, stride: 500, pixel_format: Bgr }, memory_map_addr: ed38, memory_map_len: 7 }

INFO : BIOS boot
```

Using max logging no errors or warnings it just crashes works fine without aslr

interstellarfrog, 2023-07-04:
after more testing this only seems to work for UEFI

interstellarfrog, 2023-07-07:
If this is only intended for UEFI maybe specifying this in the description of the field would be helpful

Freax13, 2023-07-22:
Can you provide an example kernel which doesn't boot with ASLR? This should work fine and is not only intended for booting with UEFI. I tried enabling ASLR in your project at https://github.com/interstellarfrog/interstellar and it booted just fine.

interstellarfrog, 2023-07-22:
> Can you provide an example kernel which doesn't boot with ASLR? This should work fine and is not only intended for booting with UEFI. I tried enabling ASLR in your project at https://github.com/interstellarfrog/interstellar and it booted just fine.

@Freax13  thanks for checking this it seems to work fine now, for some reason before it only worked with UEFI probably my fault

closed 2023-07-22 by interstellarfrog