rust-osdev/bootloader #388: Add Cargo.lock into .gitignore

yvvki, 2023-07-14:
From this documentation <https://doc.rust-lang.org/cargo/guide/cargo-toml-vs-cargo-lock.html>, it's stated that
> If you’re building a non-end product, such as a rust library that other rust [packages](https://doc.rust-lang.org/cargo/appendix/glossary.html#package) will depend on, put Cargo.lock in your .gitignore.

The [GitHub gitignore convention](https://github.com/github/gitignore.git) uses it: <https://github.com/github/gitignore/blob/main/Rust.gitignore>
```gitignore
...
# Remove Cargo.lock from gitignore if creating an executable, leave it for libraries
# More information here https://doc.rust-lang.org/cargo/guide/cargo-toml-vs-cargo-lock.html
Cargo.lock
...
```

phil-opp, 2023-07-14:
Thanks for raising this issue! I'm aware of the official recommendations, but I'm still in favor of keeping the `Cargo.lock` checked in for two reasons:

1. There are plans to change the official recommendations to always check in the `Cargo.lock`. See:
  - https://github.com/rust-lang/cargo/issues/8728
  - https://github.com/rust-lang/cargo/pull/12275
2. This project consists of multiple crates that are combined together. Most of them are built as binaries. So even with the current recommendation, we should keep the `Cargo.lock` checked in.

yvvki, 2023-07-14:
All right, I'll hold off the pull request. Thanks for the review!

closed 2023-07-14 by yvvki