rust-osdev/bootloader #396: Image builder sporadically throws ENOSPC on macOS even with 72GB of host space available

kennystrawnmusic, 2023-10-22:
After having [suddenly added 1,464 lines of code to my kernel](https://github.com/kennystrawnmusic/cryptos/commit/3d738cbec53cb2ee9796b0f2418d768f5a177361) the UEFI version of this will no longer compile:

[image]

This has happened before, but figured it was just a transient issue — ah, not anymore. The instant a large change is made, the bootloader suddenly fails to copy the kernel binary onto the image. Why is this?

Freax13, 2023-10-22:
AFAICT the out of space error is not related to your hard drive but to the bootable disk image we create. We try to calculate the smallest size we need to fit all the files on the disk image and apparently those calculations are not correct.
Can you try to add a bit more padding but increasing the value calculated in this line?
https://github.com/rust-osdev/bootloader/blob/7c8e2ca63449f92cd1f7494b9cdc9fcf58b7375d/src/fat.rs#L29

kennystrawnmusic, 2023-10-22:
> AFAICT the out of space error is not related to your hard drive but to the bootable disk image we create. We try to calculate the smallest size we need to fit all the files on the disk image and [...]

Will do that when I get home, thanks.

kennystrawnmusic, 2023-10-22:
Alright, just submitted a PR to fix this

kennystrawnmusic, 2023-10-23:
I also noticed that this only happens when compiling with debug assertions enabled; file size doesn't overflow at all and compilation still occurs without any patches if I pass `--release` to Cargo.

Freax13, 2023-10-23:
> I also noticed that this only happens when compiling with debug assertions enabled; file size doesn't overflow at all and compilation still occurs without any patches if I pass `--release` to Cargo. So using that in my VS Code configuration files as a workaround.

Does the `--release` flag also affect your kernel? If so, the generated binary will have a different size, which might explain why no error occurs.

kennystrawnmusic, 2023-10-24:
> Does the `--release` flag also affect your kernel? If so, the generated binary will have a different size, which might explain why no error occurs.

It actually improves performance, so I prefer it anyway for that reason.

PR #397 (2023-10-24): Add an additional MB of space to the generated FAT partition
Increasing the image size by an additional megabyte makes the image compile again

Fixes #396