rust-osdev/bootloader #400: Boot prcess gets sometimes stuck at "TRACE: exiting boot services" when using Uefi and qemu

daniel-keitel, 2023-11-11:
When running `qemu-system-x86_64 -bios {ovmf_pure_efi_path} -drive format=raw,file={uefi_path} -smp {core_count}`
the bootloader gets sometimes stuck at:
```
INFO : UEFI bootloader started
INFO : Using framebuffer at 0xc0000000
INFO : Reading configuration from disk was successful
INFO : Trying to load ramdisk via Disk
INFO : Ramdisk not found.
TRACE: exiting boot services
```

Higher core counts for the -smp argument result in a higher likelihood of it getting stuck.
(I am not sure if it even occurred once at -smp 1) 

- ovmf-prebuilt = "0.1.0-alpha.1"
- occurred independent of used BootloaderConfig
- never encountered on hardware 
- occurred on Linux, Mac and Windows hosts
- my toy Os repo: https://github.com/daniel-keitel/Steelmind_OS

phil-opp, 2023-11-21:
Thanks for reporting! Not sure what could cause this, but I'll try to reproduce this when I have some time.

Wasabi375, 2024-09-16:
I have the same issue, so I don't think it is related to either the kernel or the configuration.
Also I tried running this repos tests using smp and got the same error.

I'll spend some time investigating this. This bug is rather annoying when trying to run automated tests.