rust-osdev/bootloader #401: Writing to the framebuffer is slow on bare metal

daniel-keitel, 2023-11-11:
Writing to the framebuffer is extremely slow on bare metal.
Setting the caching behavior of the framebuffer to write combining results in significantly higher throughput.

The framebuffer should either be setup to use write combining in the bootloader,
or it should be mentioned in the readme/migration guides.

(It took me quite some time to figure out that that was my bottle neck)

phil-opp, 2023-11-21:
Yes, setting the cache to write combining improvies the performance considerably. Mentioning it in the README and migration guide is a good idea!

I also thought about setting it by default for the framebuffer memory region, but I'm not sure how yet. The PAT seems like the more modern approach, but it requires modifying the default config to enable write-combining caching, which might lead to issues if the kernel changes it later. MTRRs seems to be the legacy way to do it, but I haven't had time to take a coser look.

What approach did you use for your kernel?

daniel-keitel, 2023-11-25:
I changed the PAT entry for write_through + cache_disabled (index 3)  to write combining; And changed the page table flags of the framebuffer to use this entry. This is no option for the bootloader since the user doesn't expect this behavior.

A better approach would be to make use of the third bit to address the pat. 
This, as far as I know, requires the framebuffer to be mapped in huge pages.