rust-osdev/bootloader #404: Memory use during build

m-mueller678, 2023-11-20:
I was trying to build bootloader on my notebook with 4GB ram. Unfortunately, it runs out of memory and becomes unbearably slow with swapping. I think this is due to the parallel build of boot stages. It would be neat to have an option to build them sequentially.

bjorn3, 2023-11-20:
You can pass `-j1` to cargo to build only a single crate at a time.

bjorn3, 2023-11-20:
> passing -j1 does not help with that.

As far as I understand the jobserver should be passed through to those cargo sub invocations using `CARGO_MAKEFLAGS` and thus `-j1` should limit the amount of processes across all those cargo instances.

m-mueller678, 2023-11-20:
I am not familiar with cargo internals, but I have observed at least three concurrent `rustc` invocations in `top`.

bjorn3, 2023-11-21:
I think I get it. I think all four cargo invocations assumed they had an implicit token from getting spawned as they normally would and thus they all thought it would be fine to spawn one rustc instance. One fix for that would be to use the jobserver crate inside the build script to limit spawning new cargo invocations based on if a jobserver token is available.