rust-osdev/bootloader #416: Build script fails to build on latest nighly with BIOS feature enabled

Spartan2909, 2024-02-06:
This is the error:
```
   Compiling rustc-std-workspace-core v1.99.0 (C:\Users\caleb\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\rustc-std-workspace-core)     
   Compiling rustc-std-workspace-core v1.99.0 (C:\Users\caleb\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\rustc-std-workspace-core)     
   Compiling rustc-std-workspace-core v1.99.0 (C:\Users\caleb\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\rustc-std-workspace-core)     
error: data-layout for target `i686-stage-3-15880981762407934716`, `e-m:e-i32:32-f80:128-n8:16:32-S128-p:32:32`, differs from LLVM target's `i386-unknown-none` default layout, `e-m:e-p:32:32-p270:32:32-p271:32:32-p272:64:64-f64:32:64-f80:32-n8:16:32-S128`

error: could not compile `rustc-std-workspace-core` (lib) due to 1 previous error
warning: build failed, waiting for other jobs to finish...
error: could not compile `core` (lib) due to 1 previous error
error: could not compile `compiler_builtins` (lib) due to 1 previous error
error: failed to compile `bootloader-x86_64-bios-stage-3 v0.11.5`, intermediate artifacts can be found at `C:\Users\<me>\AppData\Local\Temp\cargo-installZp31yL`.
To reuse those artifacts with a future compilation, set the environment variable `CARGO_TARGET_DIR` to that path.
thread 'main' panicked at C:\Users\<me>\.cargo\registry\src\index.crates.io-6f17d22bba15001f\bootloader-0.11.5\build.rs:239:9:
failed to build bios stage-3
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

I haven't narrowed down exactly when it stopped working, but it still works on `nightly-2024-01-01`.

phil-opp, 2024-02-06:
Should be fixed already in v0.11.6, see https://github.com/rust-osdev/bootloader/pull/413. Try running  `cargo update`.

Spartan2909, 2024-02-07:
It is fixed, thanks

closed 2024-02-07 by Spartan2909