rust-osdev/bootloader #417: Kernel not mapped in higher half?

tsatke, 2024-02-09:
This is my bootloader config.
```rust
pub const fn bootloader_config() -> BootloaderConfig {
    let mut config = BootloaderConfig::new_default();
    config.mappings.page_table_recursive = Some(Mapping::Dynamic);
    config.mappings.framebuffer = Mapping::Dynamic;
    config.kernel_stack_size = KERNEL_STACK_SIZE.bytes() as u64;
    config
}
```
On boot, when printing `boot_info.kernel_image_offset`, I get `0x0000_0080_0019_6e00` although I would have expected `0x0000_8000_0019_6e00` (or similar).
As I understand it, this would only leave `0x0` - `0x80_0000_0000` (512GiB) to the userspace as everything above would be kernel space?
Am I missing a config? Am I misunderstanding something?

phil-opp, 2024-02-09:
Are you compiling your kernel as position independent or as static? The bootloader maps static kernel executables at the virtual address specified in the ELF file, which you can override through a linker command (e.g. `--image-base`) or a linker script. For dynamic mappings, you can specify a lower address bound in the bootloader config via `dynamic_range_start`.

tsatke, 2024-02-09:
I'm not sure. I'm just doing `cargo build`. This is my repo: https://github.com/tsatke/devos

My rustflags:
```toml
[target.x86_64-unknown-none]
rustflags = [
    "-C", "link-arg=-z",
    "-C", "link-arg=nostart-stop-gc",
    "-g"
]
```

So this should be a PIC, right?
I've followed your blog posts a while back @phil-opp, and migrated to 0.11 following the migration guide (if that helps).

tsatke, 2024-02-14:
Can someone point me in the right direction? Is it intended that the kernel is mapped at `0x80_0000_0000`?

closed 2024-02-14 by tsatke