rust-osdev/bootloader #427: Load kernel at specific address in v0.11

jxors, 2024-02-20:
Is there any way to load the kernel at a specific address? With `v0.10` I was using `link-arg=--image-base=0x00001eecb3620000` to ensure that the kernel was always located at `0x00001eecb3620000`, but `v0.11` no longer seems to respect the offsets in the ELF file.

Freax13, 2024-02-20:
> I'm guessing it's probably because of [this code](https://github.com/rust-osdev/bootloader/blob/b394598ddface754dbb3866c7e1498d9b4a41c95/common/src/load_kernel.rs#L81). I'm happy to write a PR that adds a config option to override the address chosen there.

The line you mentioned is only active for kernel compiled as position-independent executables. If you want your kernel to be loaded at a specific address you'll have to compile it with `-C relocation-model=static`.

Freax13, 2024-02-21:
This is caused by a bug related to the large code model:
AFAICT when the large relocation model is used, all functions and global variables that contain references to other functions and global variables outside what's reachable by the small code model are put in special `.ltext`, `.lrodata` and `.ldata` sections instead the regular `.text`, `.rodata` and `.data` sections. The bootloader_api crate internally uses a special `.bootloader-config` section to tell the bootloader how to boot the kernel. It seems to me that some part of the compiler is confused by this section. My guess is that the compiler will use 32-bit relocations by default and only use 64-bit relocations when referencing something in a `.ltext`, `.lrodata`, or `.ldata` section. Since `.bootloader-config` is not one of those, it uses a 32-bit relocation, but this fails because `.bootloader-config` cannot be referenced with a 32-bit relocation.

Freax13, 2024-02-21:
I opened #428 to work around the compiler bug.

daihuasheng, 2024-07-26:
In v0.11 kernel is placed as a workspace member of the whole project. In this case, how could I pass `-Crelocation-model=static` to the kernel without affecting bootloader?

Freax13, 2024-07-26:
> In v0.11 kernel is placed as a workspace member of the whole project. In this case, how could I pass `-Crelocation-model=static` to the kernel without affecting bootloader?

See this [example](https://github.com/rust-osdev/bootloader/blob/96f58f2d78cb9d34e6f963aa14f0307bcc953fe8/Cargo.toml#L117-L125).

PR #428 (2024-03-14): avoid 32-bit relocation to __BOOTLOADER_CONFIG
As explained in my comment in #427, the compiler seems to have trouble emitting relocations for references to global variables in custom sections. For custom sections, it always emits 32-bit relocations even when a 64-bit relocation would be required. This patch works around that by never referencing the global in the custom section directly from code, but only through a pointer from another global variable in the non-custom .data section. The relocation used for the pointer in the global variable will always use a 64-bit relocation.

I tested the [... cut]