rust-osdev/bootloader #433: Error installing "bootimage" v 1.10.3

Amongasik4334, 2024-03-25:
# Error when installing bootimage, many errors appear

I'm new to the Rust language. Before that I wrote in C, C++ and C#. So sorry if I made a stupid mistake

This is what the console outputs *The entire output did not fit*:
```
error[E0412]: cannot find type `Result` in this scope
    --> C:\Users\Пользователь\.cargo\registry\src\index.crates.io-6f17d22bba15001f\serde-1.0.197\src\private\ser.rs:1132:10
     |
1132 |     ) -> Result<Self::Ok, Self::Error>
     |          ^^^^^^ not found in this scope
     |
help: consider importing one of these items
     |
1    + use core::fmt::Result;
     |
1    + use core::result::Result;
     |

error[E0531]: cannot find tuple struct or tuple variant `Ok` in this scope
    --> C:\Users\Пользователь\.cargo\registry\src\index.crates.io-6f17d22bba15001f\serde-1.0.197\src\lib.rs:286:13
     |
286  |             Ok(val) => val,
     |             ^^ not found in this scope
     |
    ::: C:\Users\Пользователь\.cargo\registry\src\index.crates.io-6f17d22bba15001f\serde-1.0.197\src\private\ser.rs:1136:9
     |
1136 |         tri!(self.0.serialize_key(variant));
     |         ----------------------------------- in this macro invocation
     |
     = note: this error originates in the macro `tri` (in Nightly builds, run with -Z macro-backtrace for more info)
help: consider importing this tuple variant
    -->  C:\Users\Пользователь\.cargo\registry\src\index.crates.io-6f17d22bba15001f\serde-1.0.197\src\private\ser.rs:1:1
     |
1    + use core::result::Result::Ok;
     |

error[E0531]: cannot find tuple struct or tuple variant `Err` in this scope
    --> C:\Users\Пользователь\.cargo\registry\src\index.crates.io-6f17d22bba15001f\serde-1.0.197\src\lib.rs:287:13
     |
287  |             Err(err) => return Err(err),
     |             ^^^ not found in this scope
     |
    ::: C:\Users\Пользователь\.cargo\registry\src\index.crates.io-6f17d22bba15001f\serde-1.0.197\src\private\ser.rs:1136:9
     |
1136 |         tri!(self.0.serialize_key(variant));
     |         ----------------------------------- in this macro invocation
     |
     = note: this error originates in the macro `tri` (in Nightly builds, run with -Z macro-backtrace for more info)
help: consider importing this tuple variant
    -->  C:\Users\Пользователь\.cargo\registry\src\index.crates.io-6f17d22bba15001f\serde-1.0.197\src\private\ser.rs:1:1
     |
1    + use core::result::Result::Err;
     |

error[E0425]: cannot find function, tuple struct or tuple variant `Err` in this scope
    --> C:\Users\Пользователь\.cargo\registry\src\index.crates.io-6f17d22bba15001f\serde-1.0.197\src\lib.rs:287:32
     |
287  |             Err(err) => return Err(err),
     |                                ^^^
     |
    ::: C:\Users\Пользователь\.cargo\registry\src\index.crates.io-6f17d22bba15001f\serde-1.0.197\src\private\ser.rs:1136:9
     |
1136 |         tri!(self.0.serialize_key(variant));
     |         ----------------------------------- in this macro invocation
     |
     = note: this error originates in the macro `tri` (in Nightly builds, run with -Z macro-backtrace for more info)
help: a local variable with a similar name exists
     |
287  |             Err(err) => return err(err),
[... 883 lines cut ...]
   |                   ^^^^^^^^^^^^^^^^^^^^^^

error: requires `sized` lang_item
   --> C:\Users\Пользователь\.cargo\registry\src\index.crates.io-6f17d22bba15001f\serde-1.0.197\src\ser\impls.rs:810:30
    |
810 | fn format_u8(mut n: u8, out: &mut [u8]) -> usize {
    |                              ^^^^^^^^^

error: requires `sized` lang_item
    --> C:\Users\Пользователь\.cargo\registry\src\index.crates.io-6f17d22bba15001f\serde-1.0.197\src\private\de.rs:2825:17
     |
2825 |     recognized: &[&str],
     |                 ^^^^^^^

error: requires `sized` lang_item
  --> C:\Users\Пользователь\.cargo\registry\src\index.crates.io-6f17d22bba15001f\serde-1.0.197\src\private\mod.rs:30:35
   |
30 |     pub fn from_utf8_lossy(bytes: &[u8]) -> Cow<str> {
   |                                   ^^^^^

Some errors have detailed explanations: E0405, E0412, E0425, E0432, E0433, E0463, E0531.
error: could not compile `serde` (lib) due to 6230 previous errors

G:\Scripts\fkernel_test>
```

Freax13, 2024-03-25:
This looks like your Rust installation is missing the Windows target. Usually, one of the first error messages should warn about that and provide instructions for installing the target. Otherwise try `rustup target add x86_64-pc-windows-msvc`.

Amongasik4334, 2024-03-25:
> This looks like your Rust installation is missing the Windows target. Usually, one of the first error messages should warn about that and provide instructions for installing the target. Otherwise try `rustup target add x86_64-pc-windows-msvc`.

Sorry, but it didn't help. I wrote `rustup target add x86_64-pc-windows-msvc` but it didn’t help.

Amongasik4334, 2024-03-25:
> Try running the `cargo install` command in a parent folder, e.g. in `G:\Scripts` or in your home directory.
> 
> Also, what is your Rust/Cargo version? Could you give us the output of `rustc --version`?

rustc 1.79.0-nightly (2f090c30d 2024-03-23)

G:\Scripts\fkernel_test>cargo install
error: Using `cargo install` to install the binaries from the package in current working directory is no longer supported, use `cargo install --path .` instead. Use `cargo build` if you want to simply build the package.

phil-opp, 2024-03-25:
Thanks!

I meant running `cargo install bootimage` from e.g. `G:\Scripts` or `G:\`.

Amongasik4334, 2024-03-25:
> Thanks!
> 
> I meant running `cargo install bootimage` from e.g. `G:\Scripts` or `G:\`.

G:\Scripts\fkernel_test>cargo install bootimage
    Updating crates.io index
  Installing bootimage v0.10.3
    Updating crates.io index
   Compiling compiler_builtins v0.1.108
   Compiling core v0.0.0 (C:\Users\Пользователь\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\core)
   Compiling proc-macro2 v1.0.79
   Compiling unicode-ident v1.0.12
   Compiling serde v1.0.197
   Compiling serde_json v1.0.114
   Compiling anyhow v1.0.81
   Compiling thiserror v1.0.58
   Compiling quote v1.0.35
   Compiling syn v2.0.55
   Compiling serde_derive v1.0.197
   Compiling thiserror-impl v1.0.58
   Compiling rustc-std-workspace-core v1.99.0 (C:\Users\Пользователь\.rustup\toolchains\nightly-x86_64-pc-windows-msvc\lib\rustlib\src\rust\library\rustc-std-workspace-core)
   Compiling semver-parser v0.7.0
   Compiling itoa v1.0.10
   Compiling ryu v1.0.17
   Compiling json v0.12.4
   Compiling wait-timeout v0.2.0
   Compiling llvm-tools v0.1.1
error[E0463]: can't find crate for `std`

then begins what I indicated in the topic

bjorn3, 2024-03-25:
You are running it from `G:\Scripts\fkernel_test` rather than `G:\Scripts` or `G:\`.

Amongasik4334, 2024-03-25:
> You are running it from `G:\Scripts\fkernel_test` rather than `G:\Scripts` or `G:\`.

Everything is the same as `G:\`, so is `G:\Scripts`

phil-opp, 2024-03-25:
Did you put the default target setting in `C:\Users\Пользователь\.cargo\config.toml` by any chance?

The setting should go into `G:\Scripts\fkernel_test\.cargo\config.toml`, otherwise it applies to the whole system.

closed 2024-03-27 by Amongasik4334