rust-osdev/bootloader #434: Feature Request: Enable support for executing arbitrary functions in real mode

a-usr, 2024-04-24:
Sometimes it is necessary to be able to execute code in real mode, especially if you want / need to use a BIOS function that is only available in real mode. (e.g. the APM Installation check function (God why did I even decide that I needed to support this piece of legacy junk)). 

Thats why I propose to add a new field to the BootloaderConfig struct in the bootloader-api, wich takes a FnOnce, which in return is executed some time before the switch to protected mode happens.

bjorn3, 2024-04-24:
> [This](https://forum.osdev.org/viewtopic.php?p=339967&sid=736d8fc802e9a8b7b881c420ea2f05ae#p339967) Post in osdev claims that V86 mode is nonexisted when in 64 bit mode (which i think is long mode?)

Right. By the way why are you using an APM function with x86_64? Don't all x86_64 systems support ACPI? And if you can't interact with APM after booting anyway, why do you need to interact with it during booting?

a-usr, 2024-04-25:
> By the way why are you using an APM function with x86_64? 

I really just wanted a way to shutdown the system even if there was no acpi table present or there was any other reason why i couldnt interact with the acpi

> Don't all x86_64 systems support ACPI?

I really hope this is true, but i cant find anything that prooves or disprooves this statement

> And if you can't interact with APM after booting anyway, why do you need to interact with it during booting?

Well, I CAN interact with the apm after booting, since it has a protected mode 32 bit interface (of which im just noticing that it mignt not exactly be available when compiling for x86_64). The installation check however is required since 
1. Newer hardware doesnt support the APM anymore, and
2. the Installation check is the only way to find out the version and flags of the installed APM
And of course, the Installation check is only available when in real mode

a-usr, 2024-04-25:
Hmmmm
Considering acpi is a thing since 1996 I think its rather safe to assume that the table is existent in ram and usable...

a-usr, 2024-04-25:
I'll close this for now. Thank you a lot for your help!

closed 2024-04-25 by a-usr