rust-osdev/bootloader #441: Can't compile the bootloader

Milk-Cool, 2024-05-12:
Hi!

The error is:
```
error[E0463]: can't find crate for `std`
  |
  = note: the `x86_64-lowos` target may not support the standard library
  = note: `std` is required by `ovmf_prebuilt` because it does not declare `#![no_std]`
  = help: consider building the standard library from source with `cargo build -Zbuild-std`
```
The kernel is pretty much `blog_os` which you are probably already familiar with.

`/Cargo.toml`:
```toml
[package]
name = "lowos-wrapper"
version = "0.1.0"

[build-dependencies]
bootloader = "0.11"
lowos = { path = "lowos", artifact = "bin", target = "./lowos/x86_64-lowos.json" }

[dependencies]
ovmf-prebuilt = "0.1.0-alpha.1"

[workspace]
members = ["lowos"]
```
`/.cargo/config.toml`
```toml
[unstable]
# enable the unstable artifact-dependencies feature, see
# https://doc.rust-lang.org/nightly/cargo/reference/unstable.html#artifact-dependencies
bindeps = true
build-std-features = ["compiler-builtins-mem"]
build-std = ["core", "compiler_builtins"]

[build]
target = "lowos/x86_64-lowos.json"

[target.'cfg(target_os = "none")']
runner = "bootimage runner"
```
`/lowos/.cargo/config.toml`
```toml
[unstable]
build-std-features = ["compiler-builtins-mem"]
build-std = ["core", "compiler_builtins"]

[build]
target = "x86_64-lowos.json"

[target.'cfg(target_os = "none")']
runner = "bootimage runner"
```
Other files are just like in the template.

System: Ubuntu 23.10

Freax13, 2024-05-12:
Try removing the `target` key from `.cargo/config.toml`. You only want to compile `lowos` using your target, you don't want `lowos-wrapper` and its dependencies (including `ovmf_prebuilt`) to use this target.

Milk-Cool, 2024-05-12:
Yes, but here it says that a target is needed to recompile `std`. 
```
error: -Zbuild-std requires --target
```
If I don't recompile `std`, then it won't install some required packages with this error:
```
error[E0463]: can't find crate for `core`
...
error[E0463]: can't find crate for `std`
```

Freax13, 2024-05-12:
Try passing `--target x86_64-unknown-linux-gnu`. AFAICT, as of right now bindeps and custom targets (-> build-std), don't play very well together, so passing the target explicitly is required.

Depending on your use case, you might be able to use `x86_64-unknown-none` for `lowos`. `x86_64-unknown-none` doesn't need build-std.

Milk-Cool, 2024-05-12:
I want to build an image to run something that I shamelessly copy-pasted on real hardware.

Also, if I add `--target x86_64-unknown-linux-gnu` (or `--target x86_64-unknown-none`, idk what you meant to put here) to either `cargo run` or `cargo build`, it prints the following error:
```
thread 'main' panicked at src/cargo/core/compiler/unit_dependencies.rs:178:34:
no entry found for key
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

Milk-Cool, 2024-05-12:
found this: https://github.com/rust-lang/cargo/issues/10444

still open, is there maybe a workaround for this issue?

Freax13, 2024-05-12:
> found this: [rust-lang/cargo#10444](https://github.com/rust-lang/cargo/issues/10444)

Yup, I was about to say that as well.

> still open, is there maybe a workaround for this issue?

No, there isn't, but you don't need this.

You seem to have mixed the files meant to be used with older versions of the bootloader with files meant for the current version. Try the following steps:
1. Delete `/lowos/.cargo/config.toml`.
2. Replace `/.cargo/config.toml` with just
```toml
[unstable]
# enable the unstable artifact-dependencies feature, see
# https://doc.rust-lang.org/nightly/cargo/reference/unstable.html#artifact-dependencies
bindeps = true
```
3. Replace `lowos = { path = "lowos", artifact = "bin", target = "./lowos/x86_64-lowos.json" }` with `lowos = { path = "lowos", artifact = "bin", target = "x86_64-unknown-none }`

Milk-Cool, 2024-05-12:
I followed your steps, and got the following error:
```
error[E0463]: can't find crate for `std`
  |
  = note: the `x86_64-unknown-none` target may not support the standard library
  = note: `std` is required by `ovmf_prebuilt` because it does not declare `#![no_std]`
  = help: consider building the standard library from source with `cargo build -Zbuild-std`
```
So I added `build-std = ["core", "compiler_builtins"]` to `/.cargo/config.toml`. (I also tried `cargo build -Zbuild-std=core,compiler_builtins`.) Then I got
```
error: -Zbuild-std requires --target
```
So I added
```toml
[build]
target = "x86_64-unknown-none"
```
to `config.toml`, and I got the same error as before:
```
error[E0463]: can't find crate for `std`
  |
  = note: the `x86_64-unknown-none` target may not support the standard library
  = note: `std` is required by `ovmf_prebuilt` because it does not declare `#![no_std]`
  = help: consider building the standard library from source with `cargo build -Zbuild-std`
```
A little confused here...

Freax13, 2024-05-12:
Please undo

> I followed your steps, and got the following error:
>
> ```
> error[E0463]: can't find crate for `std`
>   |
>   = note: the `x86_64-unknown-none` target may not support the standard library
>   = note: `std` [...]

this

> ```
> error: -Zbuild-std requires --target
> ```
> 
> So I added
> 
> ```toml
> [build]
> target = "x86_64-unknown-none"
> ```
> 
> to `config.toml`, and I got the same error as before:

and this.

> ```
> error[E0463]: can't find crate for `std`
>   |
>   = note: the `x86_64-unknown-none` target may not support the standard library
>   = note: `std` is required by `ovmf_prebuilt` because it does not [...]

What command are you using to build/run the kernel?

Milk-Cool, 2024-05-12:
> What command are you using to build/run the kernel?
`cargo build`, though `cargo run` doesn't work either

Freax13, 2024-05-12:
Is that the whole command or are you adding additional arguments (e.g. `--target x86_64-unknown-none`)?

Milk-Cool, 2024-05-12:
Yep, that's the whole command, though I have tried `cargo build -Zbuild-std=core,compiler_builtins --target=x86_64-unknown-none`

Freax13, 2024-05-12:
That's odd. Could you repeat the content of your files?

Milk-Cool, 2024-05-12:
`/Cargo.toml`
```toml
[package]
name = "lowos-wrapper"
version = "0.1.0"

[build-dependencies]
bootloader = "0.11"
lowos = { path = "lowos", artifact = "bin", target = "x86_64-unknown-none" }

[dependencies]
ovmf-prebuilt = "0.1.0-alpha.1"

[workspace]
members = ["lowos"]
```
`/lowos/Cargo.toml`
```toml
[package]
name = "lowos"
version = "0.1.0"
edition = "2018"

[dependencies]
#bootloader = "0.9"
volatile = "0.2.6"
spin = "0.5.2"
x86_64 = "0.14.2"
pic8259 = "0.10.1"
pc-keyboard = "0.5.0"

[dependencies.lazy_static]
version = "1.0"
features = ["spin_no_std"]
```
(commented the `bootloader = "0.9"` here)
`/.cargo/config.toml`
```toml
[unstable]
# enable the unstable artifact-dependencies feature, see
# https://doc.rust-lang.org/nightly/cargo/reference/unstable.html#artifact-dependencies
bindeps = true
```
`main.rs` and `build.rs` are unchanged.

Freax13, 2024-05-12:
Have you installed the `x86_64-unknown-none` target (you can do so using `rustup target add x86_64-unknown-none`)? This doesn't fully explain the errors you sent (though they'd look almost identical), but installing the target is required in any case.

Milk-Cool, 2024-05-12:
Yes, I have installed the target:
```
user@hostname:~/Documents/GitHub/lowos-wrapper$ rustup target add x86_64-unknown-none
info: component 'rust-std' for target 'x86_64-unknown-none' is up to date
```

Freax13, 2024-05-12:
I'm very confused as to why cargo would want to build `ovmf_prebuilt` using `x86_64-unknown-none`. Could you run `cargo +nightly -Z unstable-options config get --show-origin`?

Milk-Cool, 2024-05-12:
```
unstable.bindeps = true # /home/user/Documents/GitHub/lowos-wrapper/.cargo/config.toml
# The following environment variables may affect the loaded values.
# CARGO_HOME=/home/user/.cargo
```

Milk-Cool, 2024-05-12:
also, it probably builds `ovmf_prebuilt` with `x86_64-unknown-none` because i build the whole project with that target. i'm new to rust but it's my best guess - i didn't see any config values telling that `cargo` should build it for linux

Freax13, 2024-05-12:
> also, it probably builds `ovmf_prebuilt` with `x86_64-unknown-none` because i build the whole project with that target. i'm new to rust but it's my best guess - i didn't see any config values telling that `cargo` should build it for linux

Ah, that's why I was asking you about the command you were using to build the project. Please use just `cargo build` (and **don't** use `--target`).

Milk-Cool, 2024-05-12:
Hm, that outputs
```
   Compiling bitflags v2.5.0
error[E0463]: can't find crate for `std`
  |
  = note: the `x86_64-unknown-none` target may not support the standard library
  = note: `std` is required by `bitflags` because it does not declare `#![no_std]`
  = help: consider building the standard library from source with `cargo build -Zbuild-std`
```
though when I use `cargo build -Zbuild-std=core,compiler_builtins --target=x86_64-unknown-none` it outputs more `std`-related errors:
```
error[E0463]: can't find crate for `std`
  |
  = note: the `x86_64-unknown-none` target may not support the standard library
  = note: `std` is required by `ovmf_prebuilt` because it does not declare `#![no_std]`
  = help: consider building the standard library from source with `cargo build -Zbuild-std`

error[E0463]: can't find crate for `std`
 --> /home/milan/.cargo/registry/src/index.crates.io-6f17d22bba15001f/ovmf-prebuilt-0.1.0-alpha.1/src/lib.rs:1:5
  |
1 | use std::path::{PathBuf, Path};
  |     ^^^ can't find crate
  |
  = note: the `x86_64-unknown-none` target may not support the standard library
  = help: consider building the standard library from source with `cargo build -Zbuild-std`

error[E0463]: can't find crate for `std`
  |
  = note: the `x86_64-unknown-none` target may not support the standard library
  = note: `std` is required by `bitflags` because it does not declare `#![no_std]`
  = help: consider building the standard library from source with `cargo build -Zbuild-std`
```

Freax13, 2024-05-12:
> Hm, that outputs
>
> ```
>    Compiling bitflags v2.5.0
> error[E0463]: can't find crate for `std`
>   |
>   = note: the `x86_64-unknown-none` target may not support the standard library
>   = note: `std` is [...]

That's progress though. Try adding `edition = "2021"` after `version = "0.1.0"` in `/Cargo.toml` or add `resolver = "2"` in the `workspace` section.

> though when I use `cargo build -Zbuild-std=core,compiler_builtins --target=x86_64-unknown-none` it outputs more `std`-related errors:
>
> ```
> error[E0463]: can't find crate for `std`
>   |
>   = note: the [...]

Nope, that's making things worse again.

Milk-Cool, 2024-05-12:
> Try adding `edition = "2021"`

That did work, though now a build script is causing issues:
```
error: failed to run custom build command for `lowos-wrapper v0.1.0 (/home/milan/Documents/GitHub/lowos-wrapper)`

Caused by:
  process didn't exit successfully: `/home/user/Documents/GitHub/lowos-wrapper/target/debug/build/lowos-wrapper-d1a379c54d8a204b/build-script-build` (exit status: 101)
  --- stderr
  thread 'main' panicked at build.rs:8:81:
  called `Option::unwrap()` on a `None` value
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
```

Milk-Cool, 2024-05-12:
oh, could it be an issue with `build.rs`?

Milk-Cool, 2024-05-12:
Yes, I forgot to replace `CARGO_BIN_FILE_KERNEL_kernel` with `CARGO_BIN_FILE_LOWOS_lowos`. Let me see if it works now.

Milk-Cool, 2024-05-12:
[image]

Do I need to read the documentation? I didn't see any mention of this

Freax13, 2024-05-12:
> ![image](https://private-user-images.githubusercontent.com/43724263/329863886-61038e28-4dac-49bf-b10a-8d34325e4477.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3 [...]

That's exactly what you need to do :) Here's the [relevant section](https://docs.rs/bootloader_api/0.11.7/bootloader_api/macro.entry_point.html). You'll also need to add the `bootloader_api` dependency to `/lowos/Cargo.toml`.

Milk-Cool, 2024-05-12:
It seems to be working! As far as I could tell, I need to re-implement printing to the screen, but for now I think I can close the issue. Thank you so much!

closed 2024-05-12 by Milk-Cool